FrescoLogic/FL2000 #1: make fails

voltumna, 2017-06-22:
Hello all, 
I am on ubuntu 17 and with kernel linux-headers-4.10.0-22-generic

I have followed your instructions, and went to the src/ folder, changed the Makefile so that it used 
KERNEL_PATH = /usr/src/linux-headers-4.10.0-22-generic

installed any dependencies copy/pasting the command from your readme

then I used "make" in the src directory and this is the output, any suggestions?

`make -C /usr/src/linux-headers-4.10.0-22-generic M=/home/mau/Downloads/fl2000-master/src modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-22-generic'
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_module.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_bulk.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_ioctl.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_render.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_dev.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_dongle.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_big_table.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_i2c.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_register.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_monitor.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_desc.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_interrupt.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_compression.o
  CC [M]  /home/mau/Downloads/fl2000-master/src/fl2000_surface.o
In file included from ./include/linux/thread_info.h:20:0,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:59,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:26,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
./arch/x86/include/asm/current.h:17:17: warning: passing argument 1 of ‘get_user_pages’ makes integer from pointer without a cast [-Wint-conversion]
 #define current get_current()
                 ^
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c:57:5: note: in expansion of macro ‘current’
     current,
     ^~~~~~~
In file included from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:31:0,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
./include/linux/mm.h:1293:6: note: expected ‘long unsigned int’ but argument is of type ‘struct task_struct *’
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~
In file included from ./include/linux/thread_info.h:20:0,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:59,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:26,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
./arch/x86/include/asm/current.h:17:17: warning: passing argument 2 of ‘get_user_pages’ makes integer from pointer without a cast [-Wint-conversion]
 #define current get_current()
                 ^
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c:58:5: note: in expansion of macro ‘current’
     current->mm,
     ^~~~~~~
In file included from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:31:0,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
./include/linux/mm.h:1293:6: note: expected ‘long unsigned int’ but argument is of type ‘struct mm_struct *’
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c:60:5: warning: passing argument 4 of ‘get_user_pages’ makes pointer from integer without a cast [-Wint-conversion]
[... 60 lines cut ...]
In file included from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:31:0,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
./include/linux/mm.h:1293:6: note: expected ‘struct page **’ but argument is of type ‘long unsigned int’
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c:88:18: error: too many arguments to function ‘get_user_pages’
   pages_pinned = get_user_pages(
                  ^~~~~~~~~~~~~~
In file included from /home/mau/Downloads/fl2000-master/src/fl2000_include.h:31:0,
                 from /home/mau/Downloads/fl2000-master/src/fl2000_surface.c:11:
./include/linux/mm.h:1293:6: note: declared here
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~
/home/mau/Downloads/fl2000-master/src/fl2000_surface.c:133:3: error: implicit declaration of function ‘page_cache_release’ [-Werror=implicit-function-declaration]
   page_cache_release(pages[i]);
   ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/mau/Downloads/fl2000-master/src/fl2000_surface.o' failed
make[2]: *** [/home/mau/Downloads/fl2000-master/src/fl2000_surface.o] Error 1
Makefile:1524: recipe for target '_module_/home/mau/Downloads/fl2000-master/src' failed
make[1]: *** [_module_/home/mau/Downloads/fl2000-master/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-22-generic'
Makefile:43: recipe for target 'all' failed
make: *** [all] Error 2
`

LinJiabang, 2017-06-23:
Hi , 
This issue is annoying for vendors like us. The core of the issue is that Linux kernel is fast moving target, and it causes driver compilation errors when you migrate to new kernel. As we described in the README, the driver is compiled on kernel 3.10.x . It might not compile on new kernel. We don't intend to fix this issue, unless we get strong motivation.

If you really want our support, please talk to Fresco sales team. The management team would decide whehter we should invest resource on this.

If you are a kernel developer, we believe you should have the capability to fix this issue.

closed 2017-06-26 by fresco-fl2000

voltumna, 2017-06-30:
I am simple user of linux and to be honest I find your attitude quite abrupt and negative.

fresco-fl2000, 2017-06-30:
Hi voltumna,
what you said is almost true. I might be too blunt, and I am pointing out a fact that every vendor has in common: Linux kernel supporting is painful for chip vendors because the kernel is ever changing. We are not able to keep up with the kernel changes.  To boil down the matter, it takes time as well as human resource to maintain , and time is translated to money.
Unless the company is driven by business lucrative, we probably won't do it for free.

I am sure the community is willing to help you out. There are lots of talented people in the Linux community that could fix this simple compile errors.

Good luck.

voltumna, 2017-07-13:
I agree, you guy are trying to help after all, 
well, I am checking around, hopefully I will find some way to convert the make to the latest version or something similar,

thanks and thank you for coming back to me on this

Best wishes