FrescoLogic/FL2000 #5: Error Make

rodyoukai, 2017-08-24:
Hi, 

I have this error when try command make:

rod@sereitei:~/Descargas/FL2000-master/src$ make
make -C /usr/src/linux-headers-4.12.0-041200-generic M=/home/rod/Descargas/FL2000-master/src modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.12.0-041200-generic'
  CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_module.o
In file included from /home/rod/Descargas/FL2000-master/src/fl2000_include.h:54:0,
                 from /home/rod/Descargas/FL2000-master/src/fl2000_module.c:11:
/home/rod/Descargas/FL2000-master/src/fl2000_module.c: In function ‘fl2000_disconnect’:
/home/rod/Descargas/FL2000-master/src/fl2000_module.c:236:25: error: ‘refcount_t {aka struct refcount_struct}’ has no member named ‘counter’
   dev_ctx->kref.refcount.counter);
                         ^
/home/rod/Descargas/FL2000-master/src/fl2000_log.h:44:39: note: in definition of macro ‘dbg_msg’
    printk("%s:" msg "\n", __func__, ##__VA_ARGS__);   \
                                       ^~~~~~~~~~~
scripts/Makefile.build:302: fallo en las instrucciones para el objetivo '/home/rod/Descargas/FL2000-master/src/fl2000_module.o'
make[2]: *** [/home/rod/Descargas/FL2000-master/src/fl2000_module.o] Error 1
Makefile:1512: fallo en las instrucciones para el objetivo '_module_/home/rod/Descargas/FL2000-master/src'
make[1]: *** [_module_/home/rod/Descargas/FL2000-master/src] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.12.0-041200-generic'
Makefile:44: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2

I hope you can help me.

Regards...

C-Bouthoorn, 2017-08-25:
Same error here:

```
make -C /usr/lib/modules/`uname -r`/build M=/FL2000/src modules
make[1]: Entering directory '/usr/lib/modules/4.12.8-2-ARCH/build'
  CC [M]  /FL2000/src/fl2000_module.o
In file included from /FL2000/src/fl2000_include.h:54:0,
                 from /FL2000/src/fl2000_module.c:11:
/FL2000/src/fl2000_module.c: In function ‘fl2000_disconnect’:
/FL2000/src/fl2000_module.c:236:25: error: ‘refcount_t {aka struct refcount_struct}’ has no member named ‘counter’
   dev_ctx->kref.refcount.counter);
                         ^
/FL2000/src/fl2000_log.h:44:39: note: in definition of macro ‘dbg_msg’
    printk("%s:" msg "\n", __func__, ##__VA_ARGS__);   \
                                       ^~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:303: /FL2000/src/fl2000_module.o] Error 1
make[1]: *** [Makefile:1515: _module_/FL2000/src] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.12.8-2-ARCH/build'
make: *** [Makefile:46: all] Error 2
```

I used `KERNEL_PATH = /usr/lib/modules/`uname -r`/build` in the Makefile.

`uname -r` gives `4.12.8-2-ARCH`

I have the `linux-headers` package installed (version 4.12.8-2)

midaszhou, 2017-09-03:
try  **_dev_ctx->kref.refcount.refs_**  instead of  **_dev_ctx->kref.refcount.counter_**

rodyoukai, 2017-09-03:
Hi @midaszhou , I try your advise and works, but I get this now:

> make -C /usr/src/linux-headers-4.12.0-041200-generic M=/home/rod/Descargas/FL2000-master/src 
> modules
> make[1]: se entra en el directorio '/usr/src/linux-headers-4.12.0-041200-generic'
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_module.o
> In file included from /home/rod/Descargas/FL2000-master/src/fl2000_include.h:54:0,
>                  from /home/rod/Descargas/FL2000-master/src/fl2000_module.c:11:
> /home/rod/Descargas/FL2000-master/src/fl2000_module.c: In function ‘fl2000_disconnect’:
> /home/rod/Descargas/FL2000-master/src/fl2000_log.h:44:11: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘atomic_t {aka struct <anonymous>}’ [-Wformat=]
>     printk("%s:" msg "\n", __func__, ##__VA_ARGS__);   \
>            ^
> /home/rod/Descargas/FL2000-master/src/fl2000_module.c:233:2: note: in expansion of macro ‘dbg_msg’
>   dbg_msg(
>   ^~~~~~~
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_bulk.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_ioctl.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_render.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_dev.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_dongle.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_big_table.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_i2c.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_register.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_monitor.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_desc.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_interrupt.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_compression.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_surface.o
> /home/rod/Descargas/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
> /home/rod/Descargas/FL2000-master/src/fl2000_surface.c:182:3: error: implicit declaration of function ‘page_cache_release’ [-Werror=implicit-function-declaration]
>    page_cache_release(pages[i]);
>    ^~~~~~~~~~~~~~~~~~
> cc1: some warnings being treated as errors
> scripts/Makefile.build:302: fallo en las instrucciones para el objetivo '/home/rod/Descargas/FL2000-master/src/fl2000_surface.o'
> make[2]: *** [/home/rod/Descargas/FL2000-master/src/fl2000_surface.o] Error 1
> Makefile:1512: fallo en las instrucciones para el objetivo '_module_/home/rod/Descargas/FL2000-master/src'
> make[1]: *** [_module_/home/rod/Descargas/FL2000-master/src] Error 2
> make[1]: se sale del directorio '/usr/src/linux-headers-4.12.0-041200-generic'
> Makefile:44: fallo en las instrucciones para el objetivo 'all'
> make: *** [all] Error 2

Can you help me?

midaszhou, 2017-09-03:
I replaced "page_cache_release(pages[i])"  with  "put_page(pages[i]) " to pass this error. ...

rodyoukai, 2017-09-03:
Now i get this:

> make -C /usr/src/linux-headers-4.12.0-041200-generic M=/home/rod/Descargas/FL2000-master/src modules
> make[1]: se entra en el directorio '/usr/src/linux-headers-4.12.0-041200-generic'
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_surface.o
>   CC [M]  /home/rod/Descargas/FL2000-master/src/fl2000_fops.o
> /home/rod/Descargas/FL2000-master/src/fl2000_fops.c:120:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
>   .fault = fl2000_vm_fault,
>            ^~~~~~~~~~~~~~~
> /home/rod/Descargas/FL2000-master/src/fl2000_fops.c:120:11: note: (near initialization for ‘fl2000_vma_ops.fault’)
> cc1: some warnings being treated as errors
> scripts/Makefile.build:302: fallo en las instrucciones para el objetivo '/home/rod/Descargas/FL2000-master/src/fl2000_fops.o'
> make[2]: *** [/home/rod/Descargas/FL2000-master/src/fl2000_fops.o] Error 1
> Makefile:1512: fallo en las instrucciones para el objetivo '_module_/home/rod/Descargas/FL2000-master/src'
> make[1]: *** [_module_/home/rod/Descargas/FL2000-master/src] Error 2
> make[1]: se sale del directorio '/usr/src/linux-headers-4.12.0-041200-generic'
> Makefile:44: fallo en las instrucciones para el objetivo 'all'
> make: *** [all] Error 2

Thanks for all your help @midaszhou

midaszhou, 2017-09-03:
@rodyoukai:
In Linux 4.11 kernel, function  fl2000_vm_fault() has only one parameter:  struct vm_fault *vmf, I just assume that we can get  struct vm_area_struct *vma just from vmf,  I didn't dig deep enough to prove that, but it seems right. 
I just substitue fllowing function definition:
        static int fl2000_vm_fault(struct vm_area_struct *vma,  struct vm_fault *vmf)
with:
        static int fl2000_vm_fault(struct vm_fault *vmf)
{
        struct vm_area_struct *vma = vmf->vma;
        ......

rodyoukai, 2017-09-03:
Hi again @midaszhou your advise works perfect, I can run make in scr folder without errors, now when tray run make in sample folder I get this:

> gcc main.c -o fltest
> main.c: In function ‘_alloc_frame’:
> main.c:320:8: warning: implicit declaration of function ‘ioctl’ [-Wimplicit-function-declaration]
>   ret = ioctl(fd, IOCTL_FL2000_TEST_ALLOC_SURFACE, phy_alloc);
>         ^~~~~
> /opt/hi3798/arm-hisiv200-linux/bin/arm-hisiv200-linux-gnueabi-gcc main.c -Xlinker -static -static-libgcc -o fltest_arm
> make: /opt/hi3798/arm-hisiv200-linux/bin/arm-hisiv200-linux-gnueabi-gcc: No se encontró el programa
> Makefile:9: fallo en las instrucciones para el objetivo 'fltest_arm'
> make: *** [fltest_arm] Error 127

Please help me!!!

midaszhou, 2017-09-03:
@rodyoukai:
 You need to add head file   <sys/ioctl.h>

midaszhou, 2017-09-09:
Finally I got half of a 640x480 size picture on screen by USB2 link, It turns out that my monitor only support 24bits color format.

rodyoukai, 2017-09-10:
Thanks @midaszhou  you are awesome...

fresco-fl2000, 2017-09-14:
I've fixed the compiler errors on kernel 4.11.0, and usb_control_msg() failure. Starting from some kernel version, the USB core discriminates against data buffer from stack (eg. local variables) when using usb_control_transfer.

The following line shows the core team decides that buffer  allocated from vmalloc() is banned when the client driver wishes to call usb_control_msg(). This is really rediculous.

http://elixir.free-electrons.com/linux/v3.18/source/drivers/usb/core/hcd.c#L1503

anyway, please try the latest code and see if it helps.

midaszhou, 2017-09-19:
@fresco-fl2000  Thanks a lot!
Would you please consider what I get as follows:
  1.   My USB-VGA gadget  is embedded with FL2000 chip.
  2.  I compile and install fl2000.ko module successfully under Linux 4.11.2, but as I run sample test program, I can only manage to show half of a 640x480 size picture on the screen, if I test 800x600 size picture, it shows no more than 1/4 of the screen. 
  3. I  tried to apply  render_buffer = system_buffer  with continguous physcial address ( in this case you also need to swap pixel data! ) ,in that way you have only one scatterlist for main_urb, so I thought  it may speed up bulk_transfer,  but only very little effect actually.      --- So it's not problem concerning number limit of scatter_gather lists for USB transactions.
  4. I  also tried to submit render_ctx->main_urb 4 times, each with 1/4 data of  a 640x480 picture,  before submitting  zero_length_urb as a token for ending a frame.  The upper screen shows 2 of 1/4 pictures exactly, and following half screen is BLACK.    ---- So bulk transaction works just fine.
  5. Since compression function is not applied currently ? so you have to use 24BIT color mode.
     
Don't  have datasheet of FL2000, so above is what  I can figure out now.  I think it's the problem of hardware set, other registers need to be set properly for FL2000 chip.   Need help!

fresco-fl2000, 2017-09-19:
Hi Midas,
Based on your feedback, I am guessing that your FL2000 device is enumerated at USB2.0 speed. FL2000 requires super speed bandwidth to work correctly. If you are sure that FL2000 is enumerated at super-speed, post the dmesg log so that we could look at it.

midaszhou, 2017-09-19:
@fresco-fl2000  Yes, My embedded board which the dongle is attached to only supports USB2 link, but I assume it can work with low resolution pictures, since the dongle works OK  attaching  to Windows system by USB2 link also, Max. resolution reaches 800x600.

fresco-fl2000, 2017-09-19:
Hi Midas,
OK. As you know, the FL2000 Linux driver is not tested on USB2.0 speed. I will work on the USB2.0 function on my spare time. If the management team decides to raise this issue as high priority ,  I will get it done at full speed.

midaszhou, 2017-09-19:
@fresco-fl2000:   
    Well understand,  Thanks for your efforts anyway.
    Just for reference:-----------------------------------------------------------------
    [  293.575309] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[  293.575338] fl2000_device_probe:streaming interface detected
[  293.582047] fl2000_device_probe:usb_dev(c4ca5400), minor_num(128), dev_name(fl2000-0) created
[  293.582344] ----- U1 U2 enabled -----
[  294.410275] ----- U1 U2 enabled -----
[  296.428813] fl2000_monitor_plugin_handler:Notify system to add monitor.
[  296.429052] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[  296.429065] fl2000_device_probe:Detect interrupt interface.
[  296.429078] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[  296.430007] usbcore: registered new interface driver fl2000

......

[  366.887530] bulk_main_completion: NOT in_riq()
[  366.887570] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.887785] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.887797] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.892793] --------------- enter bulk_main_completion ------------
[  366.892825] --------------- enter bulk_zero_length_completion ------------
[  366.892869] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.893085] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.893098] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.893488] bulk_zero_length_completion: NOT in_riq()
[  366.910395] --------------- enter bulk_main_completion ------------
[  366.910406] bulk_main_completion: NOT in_riq()
[  366.910445] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.910659] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.910671] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.915635] --------------- enter bulk_main_completion ------------
[  366.915646] bulk_main_completion: NOT in_riq()
[  366.915685] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.915900] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.915913] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.920901] --------------- enter bulk_main_completion ------------
[  366.920914] bulk_main_completion: NOT in_riq()
[  366.920950] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.921164] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.921176] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.926159] --------------- enter bulk_main_completion ------------
[  366.926180] --------------- enter bulk_zero_length_completion ------------
[  366.926218] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.926432] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.926444] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.926819] bulk_zero_length_completion: NOT in_riq()
[  366.943764] --------------- enter bulk_main_completion ------------
[  366.943778] bulk_main_completion: NOT in_riq()
[  366.943817] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.944032] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.944044] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.949003] --------------- enter bulk_main_completion ------------
[  366.949013] bulk_main_completion: NOT in_riq()
[  366.949047] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.949260] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.949272] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.954128] --------------- enter bulk_main_completion ------------
[  366.954137] bulk_main_completion: NOT in_riq()
[  366.954169] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.954380] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.954392] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.959395] --------------- enter bulk_main_completion ------------
[  366.959425] --------------- enter bulk_zero_length_completion ------------
[  366.959463] fl2000_bulk_prepare_urb:len(xe1000), nr_pages(225), start_offset(0x0)
[  366.959676] fl2000_bulk_prepare_urb:sglist[0], len = 0x1000
[  366.959688] fl2000_bulk_prepare_urb:------ num_sgs(1) transfer_buffer_length:(921600)
[  366.960068] bulk_zero_length_completion: NOT in_riq()
[  366.961040] fl2000_set_display_mode:Display information width:0 height:0.
[  366.961056] fl2000_render_stop:busy_list_count(2)
[  366.977036] --------------- enter bulk_main_completion ------------
[  366.977052] bulk_main_completion: NOT in_riq()
[  366.977068] fl2000_schedule_next_render:green_light off
[  366.982263] --------------- enter bulk_main_completion ------------
[  366.982274] bulk_main_completion: NOT in_riq()
[  366.982285] fl2000_schedule_next_render:green_light off
[  366.986217] fl2000_render_stop:waited 10 ms
[  366.986563] fl2000_ioctl_destroy_surface:handle(0xb6d02000)/user_buffer(0x49600000)/buffer_length(0xe1000)
[  366.986580] fl2000_ioctl_destroy_surface:surface(ce108000), handle(0xb6d02000)

[... cut ...] , surface_list_count(0)

[  366.986620] fl2000_surface_unmap(dev_ctx, surface)...
[  366.986662] fl2000_surface_unpin(dev_ctx, surface)...------!!!!!!-----
[  366.986680] vfree(surface->shadow_buffer)...
[  366.987134] kfree(surface)....
[  366.987198] fl2000_ioctl_test_release_surface:usr_addr(0xb6d02000)/phy_addr(0x49600000)/len(0xe1000)/num_pages(0xe1)
[  366.987833] fl2000_render_stop:busy_list_count(0)
[  366.987844] fl2000_render_stop:waited 0 ms
[  366.988132] fl2000_release:open_count(0)

midaszhou, 2017-09-21:
Switching  to COLOR_FORMAT_RGB_16_565,  now I can get a stable picture on my screen, however the resolution must be set at 640x480.   You can only see  small party of a  800x600 resolution pic, and also unstable.  I think that's because data compression is not yet applied.

closed 2018-01-29 by fresco-fl2000

Tork-336, 2023-01-23:
Hola al parecer el fallo persiste en kernel `Leaving directory '/usr/src/linux-headers-5.15.0-58-generic'` Ubuntu. lance con la opcion de la linea 38 y llego hasta aqui. Pero en la linea 41 rebienta por el comando que se ejecuta. No estoy muy seguro de lo que se desea realizar en esa linea.