FrescoLogic/FL2000 #6: build error on `fl2000_surface.c`

C-Bouthoorn, 2017-08-25:
```
/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/FL2000/src/fl2000_surface.c:182:3: error: implicit declaration of function ‘page_cache_release’; did you mean ‘pagecache_get_page’? [-Werror=implicit-function-declaration]
   page_cache_release(pages[i]);
   ^~~~~~~~~~~~~~~~~~
   pagecache_get_page
```

I used `KERNEL_PATH = /usr/lib/modules/`uname -r`/build` in the Makefile.

`uname -r` gives `4.12.8-2-ARCH`

I have the `linux-headers` package installed (version 4.12.8-2)

midaszhou, 2017-09-03:
Try to use  _**put_page(pages[i],0)**_  instead.

fresco-fl2000, 2017-09-14:
I've fixed the compiler errors on kernel 4.11.0, and usb_control_msg() failure. Starting from some kernel version, the USB core discriminates against data buffer from stack (eg. local variables) when using usb_control_transfer.

The following line shows the core team decides that buffer  allocated from vmalloc() is banned when the client driver wishes to call usb_control_msg(). This is really rediculous.

http://elixir.free-electrons.com/linux/v3.18/source/drivers/usb/core/hcd.c#L1503

anyway, please try the latest code and see if it helps.

closed 2018-01-29 by fresco-fl2000