FrescoLogic/FL2000 #7: build error on fl2000_fops.c

C-Bouthoorn, 2017-08-25:
```
/FL2000/src/fl2000_fops.c:120:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .fault = fl2000_vm_fault,
           ^~~~~~~~~~~~~~~
/FL2000/src/fl2000_fops.c:120:11: note: (near initialization for ‘fl2000_vma_ops.fault’)
```

I used `KERNEL_PATH = /usr/lib/modules/`uname -r`/build` in the Makefile.

`uname -r` gives `4.12.8-2-ARCH`

I have the `linux-headers` package installed (version 4.12.8-2)

midaszhou, 2017-09-03:
It seems function  **static int fl2000_vm_fault(struct vm_area_struct *vma, struct vm_fault *vmf)** shall be changed to:
static int fl2000_vm_fault(struct vm_fault *vmf)
{
        struct vm_area_struct *vma = vmf->vma;
...
This woks on Linux 4.11.2

fresco-fl2000, 2017-09-14:
I've fixed the compiler errors on kernel 4.11.0, and usb_control_msg() failure. Starting from some kernel version, the USB core discriminates against data buffer from stack (eg. local variables) when using usb_control_transfer.

The following line shows the core team decides that buffer  allocated from vmalloc() is banned when the client driver wishes to call usb_control_msg(). This is really rediculous.

http://elixir.free-electrons.com/linux/v3.18/source/drivers/usb/core/hcd.c#L1503

anyway, please try the latest code and see if it helps.

closed 2018-01-29 by fresco-fl2000