FrescoLogic/FL2000 #8: Pixel compression on USB2.0(HS)

Linux-Beginner, 2017-09-19:
About pixel compression on USB2.0(HS),
It would be greatly appreciated if you could explain the details.

As I looked at the sample source code,
Pixel data was being converted by fl2000_ioctl.c:pixel_swap() when USB3.0(SS).
As it is, converted data was being transmitted by bulkout transfer.

On Windows and USB 2.0(HS) connection,
When I try to analyze with USB analyzer, It looks like the pixel data is compressed and transmitted.
Do you need similar processing on Linux?

As I looked at the sample source code,
I think "fl2000_compression.c" seems to be compression processing, is it correct?
If correct, could you tell me how to use "fl2000_compression.c"?

midaszhou, 2017-09-23:
How about calling fl2000_compression_gravity(...) after swap render buffer

ulli-kroll, 2017-10-04:
> On Windows and USB 2.0(HS) connection,
When I try to analyze with USB analyzer, It looks like the pixel data is compressed and transmitted.
Do you need similar processing on Linux?

Can you test on Windows using HDMI ??

LinJiabang, 2019-07-25:
USB2.0 compression support is just committed . Please checkout the latest driver. Let us know if this works for you.