FrescoLogic/FL2000 #9: OSX support?

anderslundyo, 2017-09-21:
Hi - on work we recently bought a few, since it was sold to us with the promise of OSX compatibility. 
But after doing a little google search, we found it was not supported by mac. Anyone has a fix/can tweak this, so it can be used for mac?

alextunyk, 2018-04-11:
I also bought one with promise of OSX but it appeared only windows. Is there any plan to add support for mac?

LinJiabang, 2019-07-25:
FrescoLogic has macOS driver support .goto https://support.frescologic.com/portal/home

closed 2019-07-25 by fresco-fl2000

astute-engineering, 2021-04-20:
That page is invalid. Could you please update the latest location of the driver?

restuindrawan, 2021-05-14:
https://support.frescologic.com/portal/en/kb/articles/fl2000-mac-driver-beta-release

ableigdev, 2021-05-17:
@LinJiabang Links are totally broken

zencodex, 2021-05-19:
@fresco-fl2000 

support link is broken.
https://support.frescologic.com/portal/en/kb/articles/fl2000-mac-driver-beta-release

Infinite-8892, 2023-04-21:
Hello,
I have Ventura 13.3.1 and I originally had Windows Laptop. It worked fine. Moving to Ventura I installed the drivers I had. I saw they were noted for Big Sur and a bit below. I attempted them and It worked real bad. It did share screen, but only for a resolution lower. The default resolution is distorted and broken. Are there any supported drivers for it?

ozgurkazancci, 2024-05-27:
Hi, I'm on MacOS Monterey as well. Could you please kindly update the driver for Monterey version? This is not such a very new OS release anyway. It's already 3 year-old OS.

Regards.