FrescoLogic/FL2000 #10: Isochronous Transfer

midaszhou, 2017-10-08:
I tried to send data by isochronous transfer, thought it would be more stable and fast, as USB2 bandwith access is guaranteed. However, the result shows that's not the case.  see pic here:   http://imageshack.com/a/img923/57/2Gco9s.jpg
I  activated sb_ifc_streaming->altsetting[1],  the Max. payload for isochronous endpoint EP2(OUT) is 3x1024 bytes.  Some hardware setting must be forgotten??

sarman1998, 2017-10-08:
Max total bandwidth of USB 2 isoch  is 24.5MB/s. Bulk will run faster on USB2.  Picture only making to part way down the screen is exactly what you will see from FL2000 if there isn't enough bandwidth.

midaszhou, 2017-10-08:
@sarman1998:  Thanks for the tip.  Most embedded boards are currently equiped with USB2.0 hub controller, it seems we'll have to upgrade to USB3 link for 800x600 resolution and above.

midaszhou, 2017-10-08:
@sarman1998:  By the way, is there any possibility that we can improve the maximum size of ISO data payload from 3x1024 bytes to, say as 5x1024 bytes ?  by resetting some registers.

ulli-kroll, 2017-10-12:
Why do you want to do isochronous transfer ??
This should only used for time critical applications.
As @sarman1998  pointed out the is some max. bandwidth to grantee this.
Sometimes the HC driver will do a "hard" limitation, if you exceed this.

They are with this several solutions in the code to lower the bandwidth.
i.e. compression and 24bit to 15 bit RGB conversion.

ulli-kroll, 2017-10-13:
updated/fixed my last comment

Anyway, I think I have enough information found on the net how this chip is working.
The HDMI implementation is bad, because of the rewritten IT66212 driver.
There is some better (and more complete driver in this repository
git://github.com/AndrewDB/rk3066-kernel.git
But first the complexity has to be reduced.

midaszhou, 2017-10-15:
I put my testing coding here: https://github.com/midaszhou/mtest/tree/master/FL2000

ulli-kroll, 2017-10-16:
Your repository is a mess.
You need to add a reasonable .gitignore, this removes some files in your repo, 
showing up while compiling.

Anyway
In vanilla linux there is some similar driver, which uses modern DRM structure, this can be adapted for FL2000. Almost everything is there, the only parts are missing is HDMI *and* CEC, whjch is currently not an issue, I don't want to risen the complexity, but prior this I *must* clean the vendor driver.

Now I need to check if for small simple test programs using DRM or similar. I think there was some frambuffer emulation for userspace in DRM.

ulli-kroll, 2017-10-30:
Can some please check
https://github.com/ulli-kroll/FL2000/tree/rework-origin/new-urb-transter
I removed most/all of the USB2 limitations regarding the screen size.
Also I used some other schematic to transfer the data to the device, so it's possible to do pixel compress of data.

Currently there is **no** VGA display for 1680*¹1050, looks like the timings are wrong.
And also no HDMI for 1920p, tough