FrescoLogic/FL2000 #12: Manager desktop

SpinaNico, 2018-01-12:
hi my post is not a problem but is question for use this driver (fl2000.ko) on linux... 
with success make it and insmod this driver and i test it success, 
 but use this driver in Linux it's possible? 
your know some program for use this in "normal mode" normal mirror desktop and others? 
I understand perfectly that this repository for driver and testing on Linux.
but in addition to testing there were additions on linux?

thanks for reading

ulli-kroll, 2018-01-17:
This driver *is* only a prove of concept from Fresco Logic, with no DRM support
but anyway ...
```
[  689.874813] it66121: loading out-of-tree module taints kernel.
[  697.370489] usb 3-2.3: FL2000DX found
[  697.374250] usb 3-2.3: streaming interface detected
[  698.150886] i2c i2c-0: fl2k-ic2 probed
[  698.270953] it66121 0-004c: ITE66121 not found ret -19
[  698.276500] i2c i2c-0: IT66121 not found on 4c
[  698.282730] usb 3-2.3: fl2k_driver_load(): max flight 4, max transter 59392
[  698.291150] usb 3-2.3: drm_connector_init()
[  698.296296] usb 3-2.3: read EDID length 128
[  700.257059] open /dev/fb0 user=0 fb_info=ffff88010bd02800 count=1
[  700.258355] usb 3-2.3: fl2k_crtc_mode_set() hdisplay 1366 vdisplay 768
[  700.258405] Console: switching to colour frame buffer device 170x48
[  700.292249] usb 3-2.3: fb0: fl2k-drmfb frame buffer device
[  700.298034] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  700.304935] [drm] No driver support for vblank timestamp query.
[  700.315538] usb 3-2.3: Notify system to add monitor.
[  700.320868] [drm] Initialized fl2k 0.0.1 20120220 for 3-2.3 on minor 0
[  700.327552] [drm] Initialized fl2k on minor 0
[  700.332345] usb 3-2.3: Detect interrupt interface.
[  700.337968] usbcore: registered new interface driver fl2000
buildroot-4 ~ # lsmod 
Module                  Size  Used by    Tainted: G  
fl2k                   28672  1 
it66121                16384  0 
drm_kms_helper        122880  2 fl2k,it66121
drm                   303104  4 fl2k,it66121,drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
cfbfillrect            16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
cfbimgblt              16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
cfbcopyarea            16384  1 drm_kms_helper
fb                     86016 67 fl2k,drm_kms_helper
font                   20480  1 fb
fbdev                  16384  1 fb
buildroot-4 ~ # 

```
currently only a DRM concept with no further functionality.
The timing is a challenge work to do.

CanNuhlar, 2018-01-20:
I'm trying to add DRM using https://github.com/DisplayLink/evdi to this driver but I couldn't even get the monitor a display to static image maybe my unit is faulty, it is a little funky even on Windows.

ulli-kroll, 2018-01-20:
@CanNuhlar 
AFAIK: 
This is only managing multiple screens into one virtual screen.
So no real driver.

For the fork of FL2000 
If you testing my changes on
`FL2000/testing`
I really *don't* care, this is only a testing branch to discover the device with static images. If this thing is working on my side -> good

The output above if from my drm branch, which if (of course) in not working state.
Currently is working, so far

- usb device init
- drm init, including framebuffer. I think the kernel uses the framebuffer already for the console
as you can read here
`[  700.258405] Console: switching to colour frame buffer device 170x48`
- vga edid read trough drm
`/sys/class/drm/card*-VGA-1/edid`
- i2c device driver, for the embedded I2C/SMB controller
- it66121 (only) detection, I have some device with HDMI.

what is still missing 
vga timing control aka hsync/vsync, some parts in FL2000/testing needs a review, so I can (re)use them.
and *very important* 
framebuffer data transfer to the device.

And the last one is, as I said before a challenging work.
Because I need a big chunk of dma memory and a kernel thread to send this continuously to the device, while getting data from the framebuffer.

CanNuhlar, 2018-01-21:
@ulli-kroll Yeah of course, I wasn't expecting a fully working version just wanted to let you know. 

From the evdi page: "The Extensible Virtual Display Interface (EVDI) is a Linux® kernel module that enables management of multiple screens, allowing user-space programs to take control over what happens with the image. It is essentially a virtual display you can add, remove and receive screen updates for, in an application that uses the libevdi library." 

This is a part of the ubuntu usb displaylink vga converter driver. I've compiled it inserted the module into the kernel and it lists itself as a provider on xrandr, so I thought maybe we can use libevdi library to "redirect" desktop image to the device itself using fl2000 driver as it is.

virtualxtc, 2018-07-06:
@ulli-kroll  What exactly would be involved in 'testing' and 'reviewing'?  

Perhaps a way for users to 'dial in' the hsync/vsync settings and report back their findings would help?

I don't know much about video - how can I test fb transfer? are you just testing buffer sizes?

ulli-kroll, 2018-07-15:
My *biggest* problem is not hsync/vsync

It's data transfer and how to get this **really big** huge amount of data to the device.
The device has almost no memory to handle a full frame.

For full hd aka 1920 * 1080 you need to transfer ~ 6MByte for *each* frame
And for 60Hz this is ~ 500MByte, which is almost at the limit of UBS3

It ist possible to reduce the data
* fewer color depth pr pixel, 5 bits per color should pe possible
* compression, fresco logic uses some kind of RLE
but this is (to my knowlede) not well tested.

For testing I use 1378 * 768
but I have (currenly) no god/bad/whatever idea which convinces, to dump this into code

ulli-kroll, 2018-07-25:
FWIW:
current status on fl2k for DRM
```
buildroot ~ # grep . /sys/kernel/debug/dri/0/fl2k-frame*
/sys/kernel/debug/dri/0/fl2k-frame:112575
/sys/kernel/debug/dri/0/fl2k-framerate:29
```
Note here the debugfs output from the driver, 29 frames per second
There is actual *no* picture on the screen, because insufficient (slow) data stream.
```
buildroot ~ # lsmod 
fl2k 40960 1 - Live 0xffffffffa0055000 (O)
it66121 16384 0 - Live 0xffffffffa0102000 (O)
drm_kms_helper 126976 2 fl2k,it66121, Live 0xffffffffa00d0000
cfbcopyarea 16384 1 drm_kms_helper, Live 0xffffffffa00c9000
sysimgblt 16384 1 drm_kms_helper, Live 0xffffffffa00c2000
sysfillrect 16384 1 drm_kms_helper, Live 0xffffffffa00bb000
cfbimgblt 16384 1 drm_kms_helper, Live 0xffffffffa00b4000
syscopyarea 16384 1 drm_kms_helper, Live 0xffffffffa00ad000
cfbfillrect 16384 1 drm_kms_helper, Live 0xffffffffa00a6000
fb_sys_fops 16384 1 drm_kms_helper, Live 0xffffffffa009f000
fb 81920 67 fl2k,drm_kms_helper, Live 0xffffffffa0082000
fbdev 16384 1 fb, Live 0xffffffffa007b000
font 20480 1 fb, Live 0xffffffffa0073000
drm 315392 4 fl2k,it66121,drm_kms_helper, Live 0xffffffffa0007000
drm_panel_orientation_quirks 16384 1 drm, Live 0xffffffffa0000000

```
And for those who want to know, something crazy
https://osmocom.org/projects/osmo-fl2k/wiki
They use the FL2000 device to send SDR data

CanNuhlar, 2018-07-25:
Would you mind commiting these changes to your fork as it is?

ulli-kroll, 2018-07-29:
> Would you mind commiting these changes to your fork as it is? @ulli-kroll

If you want a driver which is
* incomplete
* don't show anything
* may loose your "homework" if the driver crashes

CanNuhlar, 2018-07-29:
I don't want a driver I just want the see its code at its current state. So I'm totally okay with all that.

prusswan, 2021-03-12:
@ulli-kroll it's probably a question for the devs but there must be a way to do it since it works on Windows

ulli-kroll, 2021-03-12:
@prusswan 

In which way ?
with VGA DVI or HDMI output
what kind of hardware/controller/SoC ?

This chip supports natively VGA output.
DVI and HDMI support, is only slapped via a 24bit RGB to DVI/HDMI converter.
attached to 24bit parallel output

Also the I2C is aligned to 32bit wide **data** transfers for read and write.
The latter one can overwrite some status registers in the DVI/HDMI converters .

Some small note about I2C
I2C itself has one address/command byte and at minimum one data byte for read/write.
As you can imagine you have with one,two and three data bytes padding at the end.

So you must readout the EDID (around 1024 bytes) from HDMI/DVI  trough **one** 8bit register 
multiple times in a row. You can assume here some errors in the EDID.
To circumvent the wrong alignment they patched the HDMI/DVI EDID after reading.
For VGA the 32bit alignment is negligible. 

Other thing 
The device relies heavenly on a good and fast USB host controler memory interface. 
The internal memory is small ~128kBytes I think, good for 3 or 4 USB3 bulk transfers.
Also the bulk transfers should be near isochronic, which bulk transfers are **not**
They tried to use isochronic transfers, but they are limited at around 20% to be isochronic !

I have done some tests with my APU (some AMD T40E SoC).
IIRC 
720p with 1280x720 will work, but 1080p with 1920x1080 not.
and only with VGA output due hardware bug (32bit alignment)

prusswan, 2021-03-13:
@ulli-kroll  just the advertised features as listed here http://www.frescologic.com/product/single/fl2000/

I don't understand the hardware details (but definitely useful to others who do), just that it relies on the high data rate of USB3 (the dev also mentioned this) for it to work