FrescoLogic/FL2000 #13: 800x480 resolution support

klogg, 2018-02-07:
Any chance to get values for big_table_<24,16,8>bit_r<0,1> resolution_entry structures that support 800x480 displays? I'd like to connect standard Raspberry Pi HDMI display

ulli-kroll, 2018-02-14:
I if wonder why you cant see no desktop ...
this is right, there is currently no DRM support
please read here
https://github.com/FrescoLogic/FL2000/issues/12

klogg, 2018-02-16:
@ulli-kroll yeah, I know about the DRM and I saw your "testing" branch (BTW, impressive work! hope you did not abandon it yet 😃) My problem is different - somehow driver with params I have created still outputs testing sample image incorrectly: it iget on the screen but seems like something wrong with colors or whatever, I had no time to analyze yet. I plan to switch to your codebase, it is much more clean than the original which seem to be stalled.

ulli-kroll, 2018-02-17:
With sample you the code in samplle/, right ?

Try do this with my testing branch, I've changed the USB part here. Some controllers may not happy with the scatter/gather approach in the driver especially my APU/APU2 devices form PC Engines I use for testing, some AMD Embedded SoC device.

The testing branch is used to check for some corner cases of the device, i.e. with the compression code, which can not work due some timing restrictions in the device and/or host. I've tested this barely with some bitmak (re)mapping, I've got about 100% cpu on one core with only 1280*1024.

So it's really needed to setup some shadow buffer an "stream" this to the device

klogg, 2018-02-20:
> With sample you the code in sample, right ?
Yep
> Try do this with my testing branch
Already did rebase, will try later. Also removing the whole "big_table" crap.
> So it's really needed to setup some shadow buffer an "stream" this to the device
Did you check latest changes in DRM/KMS subsystem? I think it must be much easier to create and test DRM drivers now, I think we can try it.

ulli-kroll, 2018-02-21:
I have a working DRM driver, but currently with no data transfer to the device. (*)
And this is specific to the device.
Depending of the plane layout some conversion is needed.
i.e.
on DRM with RGB 8 Bit  (with 32 bits per pixel) to 8 bit/24 RGB  windows bitmap layout.
but this is not enabled/uses yet.

Currently all I want to do, is send some generated picture data to the device to test the limitations in the area.

(*) 
for this I need the shadow buffer.

klogg, 2018-02-22:
@ulli-kroll do you have the DRM driver published somewhere?

prusswan, 2018-02-22:
For the record, the kernel source and test program is currently working on Ubuntu 18.04 Alpha 2 (linux kernel 4.13). Is there some similar driver that can be adapted to work with it? I remember there is a single usb2vga device that works on Ubuntu.

ulli-kroll, 2018-02-24:
@klogg 
As I said, I have no data transfer to the device, my **big** problem is memory.

@prusswan 

latest kernel for me is v4.14.x due some restriction in wireless work.
for DRM this may 
drivers/gpu/drm/udl/udl.ko
but this is DISPLAYLINK

some question form my side for @prusswan 
which resolution
if the picture is flickering ?

prusswan, 2018-02-24:
There were 3 resolutions detected (I will update when I have access to the machine again), shown using the _0 images since they are all flipped across the middle. No flickering...just display of static images.

ulli-kroll, 2018-03-05:
> 1920x1080 and 1280x1024 are showing the _0 bmp files.

hmm,
which output HDMI/DVI/VGA
also which CPU and/or North/Southbridge.
I get only 1280x1024 with the modified driver.

prusswan, 2018-03-06:
output is VGA using Intel HD4000
For available resolutions, it may also depend on the screen/monitor you are using..

ulli-kroll, 2018-03-07:
I have some monitor with 1920x1080, but I get only flickering with this resolution.
NO real picture, only top 50 lines or so are visible.
Running on a AMD SoC with 1GHz and most is done via DMA ...
Mabye I have some issue with the USB transfer with USB3

klogg, 2018-03-19:
@ulli-kroll have you seen this series? https://lists.freedesktop.org/archives/dri-devel/2017-September/152115.html

klogg, 2020-05-05:
Managed to get PLL setting for 800x480 by sniffing windows driver.

closed 2020-05-05 by klogg