FrescoLogic/FL2000 #14: Arch Linux 4.15.2-2

finsor, 2018-02-26:
Hi :)  
I run the build that's in the title. I successfully compiled the ko and the tests.  
`sudo ./fltest 0` recognizes the device:  
`found (1024x768) @ 60 fps` (wrong resolution - 1920x1080 - but still something)  
  
but nothing is displayed on the screen.  
  
How do I make it work?

linux-mate, 2018-08-15:
Mine does not even recognize resolution and does not see USB3-HDMI device.

https://github.com/FrescoLogic/FL2000/issues/20

Korynkai, 2021-01-24:
Kernel issue in #56 and #59 also affects Archlinux (and probably other distros using the 5.9.x kernel). Thought I would mention this.

(@FrescoLogic Why is this module not being developed to be merged into mainline at https://git.kernel.org/ ??? Or is the plan to do so as this module matures? I am quite certain communication of internal structure changes is more readily available on the kernel mailing lists, and doing so is key to seamless integration...)