FrescoLogic/FL2000 #15: Error during make

pregnolato, 2018-03-01:
valerio@tron:~/Downloads/FL2000-master/src$ sudo make
make -C /usr/src/linux-headers-4.4.0-116-generic M= modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-116-generic'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
make[2]: *** No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/entry/syscalls/../../include/generated/asm/syscalls_32.h'.  Stop.
arch/x86/Makefile:207: recipe for target 'archheaders' failed
make[1]: *** [archheaders] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-116-generic'
Makefile:46: recipe for target 'all' failed
make: *** [all] Error 2

Elvynia, 2018-03-07:
Hello,

I got the problem and found a solution by replacing `$(PWD)` by `$(shell pwd)` in the src/Makefile. Hope that helps ;)

pregnolato, 2018-03-07:
Hi Elvynia

I changed in the src/Makefile from $(PWD) to $(pwd), but I got a new error:

valerio@tron:~/Downloads/FL2000-master/src$ sudo make
make -C /usr/src/linux-headers-4.4.0-116-generic M= modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-116-generic'
make[2]: *** No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/entry/syscalls/../../include/generated/asm/syscalls_32.h'.  Stop.
arch/x86/Makefile:207: recipe for target 'archheaders' failed
make[1]: *** [archheaders] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-116-generic'
Makefile:46: recipe for target 'all' failed
make: *** [all] Error 2

Thank you.

Elvynia, 2018-03-07:
It's not only the case of `pwd`, you have to add `shell` to execute it as a command -> `$(shell pwd)`

pregnolato, 2018-03-20:
Hi Elvynia,

Thanks to try help me.

When I try to execute "make" statement using your tip, I got the message:

valerio@tron:~/Downloads/FL2000-master/src$ sudo make
[sudo] password for valerio: 
make -C /usr/src/linux-headers-4.4.0-116-generic M=/home/valerio/Downloads/FL2000-master/src modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-116-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-116-generic'
valerio@tron:~/Downloads/FL2000-master/src$ 

My MakeFile is:

fl2000-y := fl2000_module.o
fl2000-y += fl2000_bulk.o
fl2000-y += fl2000_ioctl.o
fl2000-y += fl2000_render.o
fl2000-y += fl2000_dev.o
fl2000-y += fl2000_dongle.o
fl2000-y += fl2000_big_table.o
fl2000-y += fl2000_i2c.o
fl2000-y += fl2000_register.o
fl2000-y += fl2000_monitor.o
fl2000-y += fl2000_desc.o
fl2000-y += fl2000_interrupt.o
fl2000-y += fl2000_compression.o
fl2000-y += fl2000_surface.o
fl2000-y += fl2000_fops.o
fl2000-y += fl2000_hdmi.o
ifdef CONFIG_USB_FL2000
obj-$(CONFIG_USB_FL2000) := fl2000.o
else
obj-m := fl2000.o
#KERNEL_PATH = ../../../kernel/linux-3.10.0-327.13.1.el7
#KERNEL_PATH = /usr/src/kernels/3.10.0-327.13.1.el7.x86_64
#KERNEL_PATH = /usr/src/kernels/`uname -r`
#KERNEL_PATH = /usr/src/linux-headers-`uname -r`
KERNEL_PATH = /usr/src/linux-headers-4.4.0-116-generic

all:
	make -C $(KERNEL_PATH) M=$(shell pwd) modules
clean:
	make -C $(KERNEL_PATH) M=$(shell pwd) clean
	rm -f Module.symvers
endif

Thank you.

Valerio

CanNuhlar, 2018-03-20:
Don't use sudo with make that may cause problems.

pregnolato, 2018-03-20:
Hi CanNuhlar.

I tried without sudo....

root@tron:/home/valerio/Downloads/FL2000-master/src# make
make -C /usr/src/linux-headers-4.4.0-116-generic M=/home/valerio/Downloads/FL2000-master/src modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-116-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-116-generic'
root@tron:/home/valerio/Downloads/FL2000-master/src# 

But I got the same message

Thank you

CanNuhlar, 2018-03-20:
It is already compiled check for fl2000.ko file on your directory. Just run `sudo insmod fl2000.ko` then run `dmesg`. You shoud see a line that says `usbcore: registered new interface driver fl2000`