FrescoLogic/FL2000 #16: support for Ubuntu 16.04 kernel 4.x?

gastonnina, 2018-03-21:
Hi people,

please update this for the next LTS Ubuntu 16.04 and in future 18.04

Cinerar, 2018-03-22:
There is user on reddit who is ready to write driver if he will be sponsored with devices
https://www.reddit.com/r/linux/comments/7165gs/fresco_logic_just_dropped_an_open_source_linux/

FrescoLogic? Cmon.

fresco-fl2000, 2018-03-23:
Hi Cinerar,
I've brought up the "free sample device" issue to the Fresco management team. Hopefully the management team would setup "FL2000 driver sponsoring" program.  After all, the sample device is cheap, and I don't think it would cost too much to sponsor with free sample devices.

Lucas

ulli-kroll, 2018-04-02:
@gastonnina 
Hi
you could try my testing branches on
https://github.com/ulli-kroll/FL2000/tree/testing-v1
https://github.com/ulli-kroll/FL2000/tree/testing-v2
but please be are there is **NO DRM** support here

phaltness, 2018-04-02:
Hi, ulli-kroll, i was playing with testing-v* branches, but - nothing seen at my hdmi display - no one pictures or desktop video stream. nothing (((. probably it was missed something. if you wish, i could provide my logs to be helpful.

ulli-kroll, 2018-04-02:
As I said no DRM
DRM -> Direct Render Manager
as "service" with you can talk to the graphics hardware via various user space programs.
I have a DRM framework on my side, but as mentioned here sometimes my big problem is memory.
Maybe I do code some hack here

phaltness, 2018-04-02:
@ulli-kroll, could you please provide your methods of DRM usage - maybe as simple "how-to" - ill try to help with it.

CanNuhlar, 2018-04-02:
@ulli-kroll Yeah would love to see your source code of your DRM work.

prusswan, 2018-04-20:
@fresco-fl2000  It shouldn't be very hard to ship one from your TW/CN factories..

Also, I have one such device and can possibly connect it to VMware as part of a remote testing setup. However, I don't know how feasible it is to access the display output without an actual monitor.

fresco-fl2000, 2018-04-20:
Hi prusswan, and those who are interested in getting FL2000 device samples,
I am not in position to make any promise. I would suggest you to go through Fresco Logic Sales team or marketing team (sales@frescologic.com , marketing@frescologic.com ).

ulli-kroll, 2018-04-22:
Some update, for those who can understand.
I have currently (barely) no time, busy at my main job

```
[  107.176938] fl2k: loading out-of-tree module taints kernel.
[  107.186122] usb 3-2.3: FL2000DX found
[  107.189955] usb 3-2.3: streaming interface detected
[  107.991066] i2c i2c-0: fl2k-ic2 probed
[  107.995556] i2c i2c-0: IT66121 not found on 4c
[  108.000274] usb 3-2.3: fl2k alloc shadow buffer
[  108.005054] cma: cma_alloc(cma 0000000074c44d54, count 1519, align 8)
[  108.005655] cma: cma_alloc(): returned 000000003c9399fa
[  108.010113] usb 3-2.3: mapped shadow buffer 6220800 bytes 0000000012de4ac5 -> 000000003b6f554e
[  108.029856] usb 3-2.3: drm_connector_init()
[  108.035420] usb 3-2.3: read EDID length 128
[  110.047572] open /dev/fb0 user=0 fb_info=00000000c18ed97c count=1
[  110.048942] usb 3-2.3: fl2k_crtc_mode_set() hdisplay 1366 vdisplay 768
[  110.048960] Console: switching to colour frame buffer device 170x48
[  110.083580] usb 3-2.3: fb0: fl2k-drmfb frame buffer device
[  110.089297] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  110.096225] [drm] No driver support for vblank timestamp query.
[  110.106576] usb 3-2.3: Notify system to add monitor.
[  110.111804] [drm] Initialized fl2k 0.0.1 20120220 for 3-2.3 on minor 0
[  110.118458] [drm] Initialized fl2k on minor 0

buildroot-4 ~ # cat /proc/cmdline 
BOOT_IMAGE=/boot/bzImage.64bit console=ttyS0,115200n8 cma=20M

buildroot-4 ~ # zcat /proc/config.gz | grep CMA
CONFIG_CMA=y
CONFIG_CMA_DEBUG=y
CONFIG_CMA_DEBUGFS=y
CONFIG_CMA_AREAS=7
CONFIG_DMA_CMA=y
CONFIG_CMA_SIZE_MBYTES=0
CONFIG_CMA_SIZE_SEL_MBYTES=y
CONFIG_CMA_ALIGNMENT=8
```

You need DMA CMA
No HDMI, because the I2C device is crazy
and of course no actual screen

fresco-fl2000, 2018-04-23:
For those who are interested in free samples:

I got instructions from management team that we don't have sample adapters for give-away. FrescoLogic only sells chip IC, instead of adapters.

prusswan, 2018-04-23:
@fresco-fl2000 then get one from your downstream clients, or write the drivers yourself..

davidhscope, 2018-10-19:
Dear,

I had bought a device with fl2000 buy, i use ubuntu18.04 and i can't get display running. Please help me!!!

CanNuhlar, 2018-10-19:
It's because there is no drm support yet. From a user's point of view you
may say it's unsupported at the moment.

fabiusks, 2018-10-20:
Cmooonn!! Ubuntu support is commodity :(

robmro27, 2018-10-21:
Hey I am also interested on run this on Ubuntu > 14. No open source driver ? :( I think i need to return this I don't want to downgrade Ubuntu oraz playing without Kernel to run this device. No body can help ?

mbaric, 2018-11-08:
Hey guys, I also have Ubuntu 18.04 and my 2nd shiny monitor keeps sitting in the dark for this to come live. Brrr

tuxedo0801, 2018-11-10:
+1 for a working linu display driver for FL2000 ...

lironmo, 2018-11-14:
+1 for a working linux display driver for FL2000 ...

sim1234, 2019-01-04:
cmon

Leocardoso94, 2019-02-09:
news?

Ni55aN, 2019-04-23:
Do anyone know a working adapters (USB -> VGA) for Ubuntu 18?

CalColson, 2019-06-04:
Any update? Thinking about returning this and getting a displaylink adapter

kishan93, 2019-06-14:
they got so long time yet no linux support yet ?

simon511000, 2019-07-09:
Up!

bsbeats, 2019-07-25:
I'm waiting too +1!!

OrlandoRomo, 2019-07-27:
:(

grodriguezl, 2019-09-25:
Nothing?

bsbeats, 2019-09-25:
I give up to wait and bought in usa a 4k usb adapter and now working very well in ubuntu 18!

https://www.diamondmm.com/product/bvu5500h-display-adapter/

probonopd, 2019-09-26:
That one is using chips and drivers from https://www.displaylink.com/.

grodriguezl, 2019-11-07:
I bought the adapter @bsbeats suggested (thanks), it works like a charm on ubuntu 18

bsbeats, 2019-11-07:
Haha i really appreciate to help someone! Regards @grodriguezl !

Mhowser, 2020-01-06:
Any progress on this?

msangel, 2020-01-16:
Cheap adapter with this chip is ‘Ewell USB3.0 to HDMI (EW002)’

eduardobcastro, 2020-11-21:
Fortunately I've reach out to this before buying it

ovidiovazquez, 2021-02-26:
I'm not lucky as you @eduardobcastro , well... at least my wife may use it in her Win10 Laptop

MRROOX, 2021-04-22:
2021  - this is not working :bug: :bug: :bug:  :c

jrszoio, 2021-04-24:
Hello everyone, can you send me a link with the adapter to buy? USB to VGA to use on ubuntu 20.04(displaylink chipset). I found this one "https://pt.aliexpress.com/item/33022768893.html?spm=a2g0o.cart.0.0.3a6f3c00uPCK99&mp=1", but I don't know it's the best choice.

nishant-k-290195, 2022-02-05:
> Do anyone know a working adapters (USB -> VGA) for Ubuntu 18?

NOT YET