FrescoLogic/FL2000 #18: Issues with multiple displays

thesaxonedone, 2018-05-07:
When attempting to use 2 of these adapters for extending the desktop by 2 monitors, the screens flicker and dont actually display any images.   When one of them is disconnected (or disabled in the OS), the other will start to work.   It doesnt matter which one is disabled.

MatsK, 2018-05-25:
Connecting more than one FL2000DX device to the same bus is deprecated.

Ref.: https://github.com/FrescoLogic/FL2000

StefanNa, 2018-06-25:
@thesaxonedone how did you make it work with just one monitor?
I managed to run the example files and display random pictures but how can I extend my monitor?
Thanks a lot!

thesaxonedone, 2018-06-25:
For one computer it was pretty much plug and play.   I dont recall, but I believe plugging in the device created a either a temporary drive, or device storage entity that had the drivers/installer on it.   I just ran that and it worked fine (but again only for a single monitor).   I ended up returning the adapters, so I can't really speak to it further than that

sarman1998, 2018-09-22:
@thesaxonedone, It sounds like you were running Windows, not linux.

closed 2018-09-22 by sarman1998