FrescoLogic/FL2000 #21: 在Linux内核 3.0.35编译时出现的问题该如何解决

DaugtherLoveTomato, 2018-08-29:
版主好！感谢分享。尽管要求内核版本在3.10以上，但目前我们使用的核心板内核版本为3.0.35，所以试图在该内核版本上编译：
遇到如下问题：
1、fl2000_ioctl.c ：500行，vm_mmap函数未声明；
2、fl2000_ioctl.c : 552行  vm_munmap 函数未声明；
3、fl2000_interrupt.c 151行 drain_workqueue函数未声明
4、fl2000_fops.c 158行  VM_DONTDUMP 未定义

请问有对于这些函数有没有可以替代的函数或者有其他解决办法？ 谢谢！期待您的回复。

DaugtherLoveTomato, 2018-08-29:
在不升级内核的情况下该如何解决？

Adrianf1980, 2019-05-22:
How about English ?

Ahacad, 2020-03-06:
Translations:
# Linux Kernel 3.0.35 Problems in compiling:
Hello, thanks for sharing. Though the requirement is that the kernel version should be >= 3.10, we are currently using the kernel 3.0.35, when we tried compiling on this version these problems occurred:
1、fl2000_ioctl.c ：line-500，vm_mmap function not declared；
2、fl2000_ioctl.c : line-552 vm_munmap function not declared；
3、fl2000_interrupt.c line-151 drain_workqueue function not declared
4、fl2000_fops.c  line-158 VM_DONTDUMP not defined

Are there functions to substitute them or other solutions? Thank you! Waiting eagerly for your response.

ricardojlrufino, 2021-04-21:
Related: https://github.com/FrescoLogic/FL2000/issues/65