FrescoLogic/FL2000 #22: Bad license

lkundrak, 2018-09-21:
```
// fl2000_module.c
//
// (c)Copyright 2017, Fresco Logic, Incorporated.
//
// The contents of this file are property of Fresco Logic, Incorporated and are strictly protected
// by Non Disclosure Agreements. Distribution in any form to unauthorized parties is strictly prohibited.
```

Had to stop reading right here.

Please consider licensing the code under a free-software license instead. Otherwise it does a disservice to whoever would wish tho a proper DRM driver to Linux.

A good choice that's favored by display driver writers seems to be the MIT license:
https://opensource.org/licenses/MIT

Thanks!

CanNuhlar, 2018-09-21:
There is a PR about this https://github.com/FrescoLogic/FL2000/pull/11 but never merged.

sarman1998, 2018-09-22:
CanNuhlar is correct, the license should be GPL, to match the top level license info.  I've merged the PR for this.

closed 2018-09-22 by sarman1998

lkundrak, 2018-09-24:
@sarman1998 thanks for doing this, but lack of license still doesn't indicate that the code is free software. You need to specify what license *is* actually used.

If you really intend to license this under GPL, please include a clear statement that it is the case (and we don't need to fear the lawyers knocking out our door). It is a good idea to just do what the GPL text that you already include suggests [1] and include a text like this in the README:

```
    Copyright (C) {year}  {fullname}

    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License along
    with this program; if not, write to the Free Software Foundation, Inc.,
    51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
```
[1] https://github.com/FrescoLogic/FL2000/blob/master/LICENSE#L282

In addition to that, mark the files that are covered by it clearly. That can be done using SPDX tags at the beginning of the source files, such as this one (corresponding to the license above):

```
// SPDX-License-Identifier: GPL-2.0-or-later
```

Also, this sort of thing seems to suggest that I shouldn't be reading the file in question.

```
# NOTE: DO NOT SEND THIS FILE OUTSIDE OF FRESCO LOGIC.
```

(I'm not sending in a pull request, since it's the copyright holder that needs to declare the license)

Thank you