FrescoLogic/FL2000 #25: sample, make error

fahri314, 2019-01-08:
insmod fl2000.ko command working but in sample directory make step gives the following error:

```cd sample/
root@debian:~/FL2000/sample# make 
/opt/hi3798/arm-hisiv200-linux/bin/arm-hisiv200-linux-gnueabi-gcc main.c -Xlinker -static -static-libgcc -o fltest_arm
make: /opt/hi3798/arm-hisiv200-linux/bin/arm-hisiv200-linux-gnueabi-gcc: Command not found
Makefile:9: recipe for target 'fltest_arm' failed
make: *** [fltest_arm] Error 127
root@debian:~/FL2000/sample# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch```

cahz, 2019-01-10:
There is a second make target for cross compiling to arm. So either you ignore the error or run `make fltest` only.
In both cases there should be an executable and you can run the test  `./fltest 0` as superuser.

fahri314, 2019-01-11:
I was able to run the test. the image did not come to the screen. it does not see the device right now
`fl2000 device not connected?
`

cahz, 2019-01-11:
try to reboot and reload the kernel module with the fl2000 device and a monitor attached

fahri314, 2019-01-11:
`user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 0
[sudo] password for user: 
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
create_surface(957, 1919) , type(0x0)
IOCTL_FL2000_SET_DISPLAY_MODE failed -1
create_surface(1920, 1080) , type(0x0)
IOCTL_FL2000_CREATE_SURFACE failed -1
create_surface(1024, 768) , type(0x0)
IOCTL_FL2000_CREATE_SURFACE failed -1
user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 1
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
create_surface(957, 1919) , type(0x1)
IOCTL_FL2000_SET_DISPLAY_MODE failed -1
create_surface(1920, 1080) , type(0x1)
IOCTL_FL2000_CREATE_SURFACE failed -1
create_surface(1024, 768) , type(0x1)
IOCTL_FL2000_CREATE_SURFACE failed -1
user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 2
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
usr_addr(0x7f13c9ba3000), phy_addr(0x15e400000) allocated
create_surface(1024, 768) , type(0x2)
display_mode(1024, 768), press any key to continue

user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 0
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
create_surface(957, 1919) , type(0x0)
IOCTL_FL2000_SET_DISPLAY_MODE failed -1
create_surface(1920, 1080) , type(0x0)
IOCTL_FL2000_CREATE_SURFACE failed -1
create_surface(1024, 768) , type(0x0)
IOCTL_FL2000_CREATE_SURFACE failed -1
user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 2
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
usr_addr(0x7f27bc384000), phy_addr(0x15e400000) allocated
create_surface(1024, 768) , type(0x2)
display_mode(1024, 768), press any key to continue

user@debian:~/Downloads/FL2000/sample$ sudo ./fltest 3
found (1024, 768) @ 60 fps
entry [0] = (1024, 768) added
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1024, 768) @ 60 fps
found (1920, 1080) detailed timing desc
entry [1] = (1920, 1080) added
found (1920, 1080) detailed timing desc
found (1920, 1080) detailed timing desc
found (957, 1919) detailed timing desc
entry [2] = (957, 1919) added
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
_alloc_frame: IOCTL_FL2000_TEST_ALLOC_SURFACE failed -1
usr_addr(0x7f4f8f264000), phy_addr(0x15e400000) allocated
create_surface(1024, 768) , type(0x3)
display_mode(1024, 768), press any key to continue
`

bhuvansingla, 2020-02-02:
@fahri314 I'm also facing this. Did you arrive at a solution?

fahri314, 2020-02-03:
No. I dont use it anymore

UntwistedApple, 2020-12-02:
In main.c the hardcoded the name of the driver file in /dev/, you can check what the name of the file in your /dev/ is and change the hardcoded name in main.c.
The line looks like `#define	FL2K_NAME	"/dev/fl2000-0"`
This kinda worked for me, but I am struggling with a different error now..

diabluf, 2021-08-29:
Hey could you tell me how you make it work please, I do get this 
[image]