FrescoLogic/FL2000 #27: I'm getting an error when the device plugged in

jamesjallorina, 2019-03-09:
[  677.053384] fl2000_dongle_card_initialize:ITE hdmi chip powered up
[  677.829477] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
[  677.829640] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[  677.829642] fl2000_device_probe:Detect interrupt interface.
[  677.829644] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[  693.087211] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
[  693.363202] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event

TheoMornico, 2019-07-10:
Hello,

I have the same error than you, did you find any solution ?

thanks

jamesjallorina, 2019-07-11:
hmm I'm don't have the chance to debug the kernel maybe this weekend I'll try to fix it.

Let me know if you have fixed it.

Thanks!

pinedamg, 2020-07-08:
Hi, I'm experiencing the same issue, anyone was able to solve?
Please if someone had a chance to give me a clue at least it would be great.

Thanks

Distro: Ubuntu 20.04
Kernel: 5.4.0-40-generic

`[ 1715.294524] usb 2-3: new SuperSpeed Gen 1 USB device number 14 using xhci_hcd
[ 1715.314678] usb 2-3: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[ 1715.314679] usb 2-3: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[ 1715.314679] usb 2-3: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[ 1715.314680] usb 2-3: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[ 1715.314681] usb 2-3: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[ 1715.314681] usb 2-3: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[ 1715.314682] usb 2-3: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[ 1715.314683] usb 2-3: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[ 1715.314683] usb 2-3: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[ 1715.314684] usb 2-3: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[ 1715.314684] usb 2-3: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[ 1715.314685] usb 2-3: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[ 1715.314685] usb 2-3: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[ 1715.314686] usb 2-3: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[ 1715.314688] usb 2-3: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[ 1715.314689] usb 2-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1715.315817] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[ 1715.315819] fl2000_device_probe:streaming interface detected
[ 1715.315894] fl2000_device_probe:usb_dev(00000000a418f937), minor_num(2), dev_name(fl2000-2) created
[ 1717.194604] fl2000_monitor_plugin_handler:Notify system to add monitor.
[ 1717.194733] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[ 1717.194734] fl2000_device_probe:Detect interrupt interface.
[ 1717.194735] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[ 1717.198865] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event`

artin961, 2021-07-09:
Hi, does someone resolve the issue. 

`[  691.232759] usb 10-1: new SuperSpeed Gen 1 USB device number 4 using xhci_hcd
[  691.253029] usb 10-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[  691.253033] usb 10-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[  691.253035] usb 10-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[  691.253038] usb 10-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[  691.253040] usb 10-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[  691.253043] usb 10-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[  691.253045] usb 10-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[  691.253048] usb 10-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[  691.253050] usb 10-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[  691.253052] usb 10-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[  691.253055] usb 10-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[  691.253057] usb 10-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[  691.253060] usb 10-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[  691.253062] usb 10-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[  691.253067] usb 10-1: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[  691.253069] usb 10-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  691.253900] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[  691.253902] fl2000_device_probe:streaming interface detected
[  691.254030] fl2000_device_probe:usb_dev(000000008d193ea2), minor_num(2), dev_name(fl2000-2) created
[  693.140545] fl2000_monitor_plugin_handler:Notify system to add monitor.
[  693.140641] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[  693.140643] fl2000_device_probe:Detect interrupt interface.
[  693.140644] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[  693.141488] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
`