FrescoLogic/FL2000 #29: Mainline?

mxmilkb, 2019-05-01:
Any possibility of getting the driver code into the mainline? Which would allow newer versions of Linux to use the hardware automatically after that.

mk-pmb, 2019-07-09:
Looks like we'll need to clean up some of the permissions notes, at least the Makefile directly contradicts the license:

```text
$ grep -nFe OUTSIDE -- src/Makefile
7:# NOTE: DO NOT SEND THIS FILE OUTSIDE OF FRESCO LOGIC.
```

Also there are some todos waiting to be fixed.

```text
$ git grep -nFe TODO -- include/ src/
src/fl2000_big_table.c:5:// Purpose: TODO: PLEASE RENAME THIS FILE TO SOMETHING USEFUL
src/fl2000_compression.c:356:           // TODO:
src/fl2000_monitor.c:691:       // TODO: FL2000DX should not need this step per Stanley's description.
```

realsobek, 2023-07-20:
Please, merge this driver into Linux mainline. Compiling custom kernels is not viable.

mk-pmb, 2023-07-21:
@realsobek
If the above legal and quality issues are fixed nowadays, you should have demonstrated that. To me, it seems they are not. From the perspective of the Linux kernel, this is unusable code. Merging it at this stage would cause the Linux team lots of trouble. You're basically suggesting an attack on Linux. Which probably nobody will even try.