FrescoLogic/FL2000 #30: BUG: unable to handle kernel NULL pointer dereference at 0000000000000100

jjakob, 2019-05-14:
OS: PureOS (debian testing based)
Kernel: 4.19.0-4-amd64

After plugging in the adapter, the kernel throws a bug in usb-storage, trace:

[69845.103182] Call Trace:
[69845.103225]  usb_stor_port_reset+0x43/0x60 [usb_storage]
[69845.103266]  usb_stor_invoke_transport+0x127/0x500 [usb_storage]
[69845.103289]  ? storage_probe+0x320/0x320 [usb_storage]
[69845.103305]  ? wait_for_completion_interruptible+0x135/0x1e0
[69845.103322]  ? wake_up_q+0x70/0x70
[69845.103344]  usb_stor_control_thread+0x1c9/0x280 [usb_storage]
[69845.103360]  ? __kthread_parkme+0x4c/0x70
[69845.103373]  kthread+0x112/0x130
[69845.103386]  ? kthread_bind+0x30/0x30
[69845.103402]  ret_from_fork+0x35/0x40

Full log attached.
[fl2000_nullptr_bug.txt](https://github.com/FrescoLogic/FL2000/files/3179120/fl2000_nullptr_bug.txt)

LinJiabang, 2019-07-25:
Hi,
We've just updated the driver, and we've found a DMA bug which might be related to the issue you described. Please check if the latest driver works. If not, let us know how to replicate the issue.

fresco-fl2000, 2019-07-26:
Um. I just looked at the log, and it seems like the USB dongle you are using has a onboard storage device. There is a known bug in the hardware design where the USB descriptors are not correctly coded such that it would cause USB storage device to be reset/dropped when FL2000 driver is loaded. I can't help on this. I believe there is no software patch.

closed 2019-07-26 by fresco-fl2000

jjakob, 2019-07-26:
Thanks, I don't need the dongle working, I was just curious how far I could get it. So it's a hardware bug? The same dongle works fine in Windows.