FrescoLogic/FL2000 #31: make no error,but run fltest "device no connect"

lvzusheng, 2019-05-28:
run "ls /dev"  can find "fl2000-3"
but run"sudo ./fltest "  " fl2000 device not connected?"
and No screen signal

dan1lo, 2019-07-16:
What is your linux? Mine is ubuntu 19.6 and don' works too

ConsciousCode, 2019-10-24:
In main.c, it has the device path hard-coded to /dev/fl2000-0, but the driver created /dev/fl2000-1. Easy fix is to just change it to the correct name, "nicer" fix would be to search for the glob /dev/fl2000-*. When I run fltest with this edit, it runs with no error, though it doesn't seem to be generating an HDMI signal.

diegodlpereira, 2020-04-30:
@ConsciousCode thanks - your proposed fix worked like a charm, nice catch

trademark007, 2022-05-25:
> In main.c, it has the device path hard-coded to `/dev/fl2000-0`, but the driver created `/dev/fl2000-1` for me. Easy fix is to just change it to the correct name, "nicer" fix would be to search for the glob `/dev/fl2000-*`. When I run `fltest` with this edit, it runs with no error, though it doesn't seem to be generating an HDMI signal.

not work for me

trademark007, 2022-05-25:
> @ConsciousCode thanks - your proposed fix worked like a charm, nice catch

any suggests?