FrescoLogic/FL2000 #32: insmod error...

andlfonseca, 2019-06-14:
➜ insmod fl2000.ko
insmod: ERROR: could not insert module fl2000.ko: Operation not permitted

➜ sudo insmod fl2000.ko
insmod: ERROR: could not insert module fl2000.ko: Required key not available

I got this problem after the fl2000 complication... I am using Linux Mint 18 with 4.15.0-43-generic.

CanNuhlar, 2019-06-14:
Hi,

Use sudo white inserting module

14 Haz 2019 Cum 23:23 tarihinde andlfonseca <notifications@github.com> şunu
yazdı:

kira2016, 2019-06-19:
https://docs.fedoraproject.org/en-US/Fedora/26/html/System_Administrators_Guide/sect-signing-kernel-modules-for-secure-boot.html
search key words “insmode  XXXX   Required key not available”