FrescoLogic/FL2000 #34: 8bit palette mode / grayscale possible?

r2x0t, 2019-06-29:
How does the hardware 8bit palette mode work?
Experimenting with the code I can enable it by setting bits 25 and 26 in REG_OFFSET_8004. That seems to work and unlike 8bit mode, where RGB is packed into single byte, I get various colors for various byte values.
Is this palette fixed or stored somewhere in the memory? Can it be changed?
I would like to run adapter in 256 grayscale mode if possible... can this be done?

Thanks.

A-Dunstan, 2025-01-13:
In case anyone's still wondering about this 6 years later...
Writing to register 805C allows setting the palette entries; the low 8 bits are the palette index to update and the upper 24 bits are the RGB values to be inserted.

r2x0t, 2025-01-14:
Thanks! I will try this to reduce USB load when generating signals (using it as cheap ADC).