FrescoLogic/FL2000 #39: Issue with Kernel 5.3

IamRichter, 2020-02-12:
So... I know that only the kernel 4 is compatible, and there is other tickets open here for the same issue, but i wanted to share my dmesg.
I had no problems compiling it, and loading, but i just did not worked...

```
[ 1012.391492] fl2000: loading out-of-tree module taints kernel.
[ 1012.391552] fl2000: module verification failed: signature and/or required key missing - tainting kernel
[ 1012.392015] usbcore: registered new interface driver fl2000  
[ 1020.115722] usb 2-1: new SuperSpeed Gen 1 USB device number 45 using xhci_hcd  
[ 1020.136065] usb 2-1: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[ 1020.136071] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1020.138128] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[ 1020.138133] fl2000_device_probe:streaming interface detected
[ 1020.138272] fl2000_device_probe:usb_dev(0000000091545f0f), minor_num(1), dev_name(fl2000-1) created
[ 1020.927915] fl2000_monitor_vga_status_handler:Ignore MonitorPlugout evethjent, monitor not attached.
[ 1020.928133] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[ 1020.928136] fl2000_device_probe:Detect interrupt interface.
[ 1020.928138] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[ 1033.596363] nouveau 0000:01:00.0: Enabling HDA controller
[ 1033.696731] nouveau 0000:01:00.0: bus: MMIO read of 00000000 FAULT at 6013d4 [ IBUS ]
[ 1036.603524] fl2000_monitor_plugin_handler:Notify system to add monitor.
```

Good luck with your projects.

fresco-fl2000, 2020-04-16:
I just worked around a FL2000 driver bug a couple minutes ago so that it works with my Fedora31 (kernel 5.5.15).  See if the latest driver helps. The major change is in fl2000_dev.c.  See the comments there.

diegodlpereira, 2020-04-30:
Hi - Got the same results as @IamRichter however it is still not working. As of @fresco-fl2000 comment above the latest change hasn't worked on my Linux 5.3.0-46-generic.

`[  190.117061] fl2000: loading out-of-tree module taints kernel.
[  190.117166] fl2000: module verification failed: signature and/or required key missing - tainting kernel
[  190.117761] usbcore: registered new interface driver fl2000
[  202.549860] usb 4-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  202.570091] usb 4-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x81, skipping
[  202.570096] usb 4-1: config 1 interface 1 altsetting 0 has a duplicate endpoint with address 0x1, skipping
[  202.570098] usb 4-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x81, skipping
[  202.570101] usb 4-1: config 1 interface 1 altsetting 1 has a duplicate endpoint with address 0x1, skipping
[  202.570104] usb 4-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x81, skipping
[  202.570107] usb 4-1: config 1 interface 1 altsetting 2 has a duplicate endpoint with address 0x1, skipping
[  202.570110] usb 4-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x81, skipping
[  202.570112] usb 4-1: config 1 interface 1 altsetting 3 has a duplicate endpoint with address 0x1, skipping
[  202.570115] usb 4-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x81, skipping
[  202.570118] usb 4-1: config 1 interface 1 altsetting 4 has a duplicate endpoint with address 0x1, skipping
[  202.570120] usb 4-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x81, skipping
[  202.570123] usb 4-1: config 1 interface 1 altsetting 5 has a duplicate endpoint with address 0x1, skipping
[  202.570126] usb 4-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x81, skipping
[  202.570128] usb 4-1: config 1 interface 1 altsetting 6 has a duplicate endpoint with address 0x1, skipping
[  202.570135] usb 4-1: New USB device found, idVendor=1d5c, idProduct=2000, bcdDevice= 2.00
[  202.570137] usb 4-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  202.571033] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[  202.571036] fl2000_device_probe:streaming interface detected
[  202.571186] fl2000_device_probe:usb_dev(00000000d867e56c), minor_num(1), dev_name(fl2000-1) created
[  204.453762] fl2000_monitor_plugin_handler:Notify system to add monitor.
[  204.453942] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[  204.453944] fl2000_device_probe:Detect interrupt interface.
[  204.453945] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[  204.454999] fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
`