FrescoLogic/FL2000 #43: three display in Manjaro

SoyDiego, 2020-05-09:
Sorry, i'm newbie here...
Hi, i have two display working perfectly (my laptop and one monitor) and i wanna use one monitor more.

I bought an USB to VGA Adapter but I don't know how can I install or setup.
This is my adapter: https://www.newegg.com/p/2VR-00D3-00042 1

Is it possible use FL2000 in Manjaro to use 3 display?

cesarcruzc, 2020-09-11:
Hey Diego,

Te funciono algo? tambien tengo linux, yo uso Pop OS!

CanNuhlar, 2020-09-11:
No, It is not possible due to this driver lacking DRM support.

SoyDiego, 2020-09-13:
> Hey Diego,
> 
> Te funciono algo? tambien tengo linux, yo uso Pop OS!

Hi Cesar, my solution was buy an HDMI to VGA adapter.
I couldn't with USB to VGA.

(Escribo en inglés por si a otra persona le sirve la respuesta :) )

closed 2020-09-13 by SoyDiego