FrescoLogic/FL2000 #46: Linux 54, xrandr shows device as disconnected.

kaiwalyajoshi, 2020-05-31:
I've compiled the driver for linux54 `5.4.40-1-MANJARO` and loaded the kernel driver via `insmod`

dmesg shows the driver is loaded:
```
fl2000: loading out-of-tree module taints kernel.
fl2000: module verification failed: signature and/or required key missing - tainting kernel
fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
fl2000_device_probe:streaming interface detected
fl2000_device_probe:usb_dev(00000000e1e4e4d2), minor_num(1), dev_name(fl2000-1) created
fl2000_dongle_card_initialize:found ITE hdmi chip, initializing it.
fl2000_dongle_card_initialize:ITE hdmi chip powered up
fl2000_monitor_read_edid:169 EDID extensions found
fl2000_monitor_plugin_handler:Notify system to add monitor.
fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
fl2000_device_probe:Detect interrupt interface.
fl2000_intr_pipe_create:found ep_num_intr_in(3)
usbcore: registered new interface driver fl2000
fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
fl2000_monitor_plugout_handler:Notify system to delete monitor.
fl2000_render_stop:busy_list_count(0)
fl2000_render_stop:waited 0 ms
fl2000_monitor_read_edid:169 EDID extensions found
fl2000_monitor_plugin_handler:Notify system to add monitor.
fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
fl2000_monitor_vga_status_handler:WARNING Ignore MonitorPlugin event
```

But `xrandr` shows the device `VIRTUAL1` as disconnected:
```
VIRTUAL1 disconnected (normal left inverted right x axis y axis)
```
What's needed to get xrandr to emit output to the device?

I have a dual boot with Windows 10 and this worked correctly there, so the device isn't defective or anything like that.

kaiwalyajoshi, 2020-05-31:
Also note that my device shows up as `/dev/fl2000-1` are any code changes required for this?

[This comment in the code mentions](https://github.com/FrescoLogic/FL2000/blob/master/src/fl2000_ioctl.h#L100-L117) this:
```
/*
 * FL2000 device presence detection
 *
 * Due to PNP nature, the FL2000 device can be unplugged during normal user app
 * operation. When a FL2000 device is disconnected, the user app must release
 * any opened file to the /dev/fl2000-0(or similar) for the driver to completely
 * releases itself. Failure to do so would cause driver lock-up.
 *
 * The following method is provided to detect FL2000 device connection.
 * 1. Polling
 *    The user app can constantly checking the existence of /dev/fl2000-0.
 *    If the FL2000 device is unplugged, the driver deregister this device name.
 * 2. Using IOCTL_FL2000_WAIT_FOR_MONITOR_EVENT.
 *    User app can issues this IOCTL in a separate thread.  The kernel driver
 *    wakes up the waiting thread, and updates fl2000_flags and monitor_flags.
 *    User app can detect monitor and FL2000 arrival/departure event by using
 *    this IOCTL.
 */
```

ftoledo, 2023-03-16:
hi @kaiwalyajoshi  What is the fork with which you obtained these results?