FrescoLogic/FL2000 #50: insmod error

Guilherme99, 2020-07-26:
insmod: ERROR: could not insert module fl2000.ko: File exists

marcocajeao, 2020-10-05:
It throws me the same error.

marcocajeao, 2020-10-05:
I need it to work on Ubuntu 20.04, but it returns an error when I run the insmod command. Please help me!

CanNuhlar, 2020-10-06:
You have to run `rmmod` first to unload module then run `insmod` again. Check the output of `lsmod` to see if it's already loaded as a kernel module

marcocajeao, 2020-10-07:
Well, I run:
`insmod fl2000.o `
and it returns this error: 
`rmmod:` ERROR: Module fl2000 is not currently loaded`
So, I run:
`rmmod fl2000.o `
and it returns: 
`rmmod: ERROR: Module fl2000 is not currently loaded`
and in `lsmod` I don't understand which module it should return.