FrescoLogic/FL2000 #51: fl2000 device not connected?

vyunak, 2020-10-09:
» lsusb                                                                                                                                                                      29 ↵ 
Bus 002 Device 004: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter

» sudo ./fltest 0                                                                                                                                                                 
fl2000 device not connected?

» sudo insmod ../src/fl2000.ko                                                                                                                                               29 ↵ 
insmod: ERROR: could not insert module ../src/fl2000.ko: File exists

work on windows 10, not work linux in ubuntu 20.04

worraps, 2020-10-21:
>  not work linux in ubuntu 20.04

confirmed - same result

dmitriy-nz, 2020-10-21:
ubuntu 18.04 same error

whydea, 2020-10-22:
In my /dev folder the file fl2000-0 was not created, why?

dmitriy-nz, 2020-10-23:
> In my /dev folder the file fl2000-0 was not created, why?

It's not a file, it's device ID in the file sample/main.c. You have to change this to your ID device.