FrescoLogic/FL2000 #59: FL2000-master/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?   109 |    down_read(&current->mm->mmap_sem);

wammachado, 2021-01-18:
make -C /usr/src/linux-headers-`uname -r` M=/home/wender/Downloads/FL2000-master/src modules
make[1]: Entrando no diretório '/usr/src/linux-headers-5.8.0-36-generic'
  CC [M]  /home/wender/Downloads/FL2000-master/src/fl2000_surface.o
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |    down_read(&current->mm->mmap_sem);
      |                            ^~~~~~~~
      |                            mmap_base
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c:115:26: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |    up_read(&current->mm->mmap_sem);
      |                          ^~~~~~~~
      |                          mmap_base
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c:129:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |   down_read(&current->mm->mmap_sem);
      |                           ^~~~~~~~
      |                           mmap_base
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c:144:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |   up_read(&current->mm->mmap_sem);
      |                         ^~~~~~~~
      |                         mmap_base
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/wender/Downloads/FL2000-master/src/fl2000_surface.c:228:28: error: too many arguments to function ‘vm_map_ram’
  228 |   surface->mapped_buffer = vm_map_ram(
      |                            ^~~~~~~~~~
In file included from /home/wender/Downloads/FL2000-master/src/fl2000_include.h:30,
                 from /home/wender/Downloads/FL2000-master/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:91:14: note: declared here
   91 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:290: /home/wender/Downloads/FL2000-master/src/fl2000_surface.o] Erro 1
make[1]: *** [Makefile:1780: /home/wender/Downloads/FL2000-master/src] Erro 2
make[1]: Saindo do diretório '/usr/src/linux-headers-5.8.0-36-generic'
make: *** [Makefile:42: all] Erro 2

freddale4, 2021-01-19:
I have the same problem on ubuntu 20.04. 
I found a thread https://github.com/snuf/iomemory-vsl4/issues/25 about this issue.

Korynkai, 2021-01-24:
Also affects Archlinux (current kernel 5.9.14-arch1-1), and probably other distros using the 5.9.x kernel.

@FrescoLogic Why is this module not being developed to be merged into mainline at https://git.kernel.org/ ??? Or is the plan to do so as this module matures? I am quite certain communication of internal structure changes is more readily available on the kernel mailing lists, and doing so is key to seamless integration...

ricodol, 2021-01-25:
Same on debian,
5.8.0-0.bpo.2-amd64

In function ‘fl2000_surface_pin_down’:
/FL2000/FL2000/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
    down_read(&current->mm->mmap_sem);

                            ^~~~~~~~

freddale4, 2021-01-28:
I get the version of [Marc-Pierre-Barbier](https://github.com/Marc-Pierre-Barbier/FL2000)...thank`s Marc!
This version compiled ok in my ubuntu 20.
I enabled Secure Boot to avoid the error "insmod: ERROR: could not insert module fl2000.ko: Operation not permitted"

Now I getting black screen in my external monitor. Just trying to figure out what is the problem now. Thoughts?

ricodol, 2021-01-28:
Compile also on debian.
[Marc-Pierre-Barbier](https://github.com/Marc-Pierre-Barbier/FL2000)...thank`s Marc!
but dmesg 

```
[ 1736.331737] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[ 1736.331739] ------------[ cut here ]------------
[ 1736.331739] refcount_t: saturated; leaking memory.
[ 1736.331750] WARNING: CPU: 1 PID: 6113 at lib/refcount.c:22 refcount_warn_saturate+0x4e/0xf0
[ 1736.331751] Modules linked in: fl2000(OE+) hidp hid_logitech_hidpp hid_jabra hid_logitech_dj cdc_mbim cdc_wdm cdc_ncm usbnet mii nls_ascii nls_cp437 vfat fat sd_mod sg uas usb_storage uhid algif_hash rfcomm xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter nft_chain_nat xt_addrtype nft_compat nf_tables nfnetlink xt_conntrack nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay ctr ccm ashmem_linux(C) binder_linux iptable_filter fuse cpufreq_powersave cpufreq_conservative cpufreq_userspace cmac bnep binfmt_misc squashfs zstd_decompress loop snd_hda_codec_hdmi snd_soc_skl snd_hda_codec_realtek snd_hda_codec_generic snd_soc_hdac_hda snd_hda_ext_core snd_soc_sst_ipc snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi snd_soc_core snd_compress snd_hda_intel snd_intel_dspcfg x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec uvcvideo kvm_intel iwlmvm btusb btrtl btbcm snd_usb_audio videobuf2_vmalloc btintel mac80211
[ 1736.331772]  videobuf2_memops snd_hda_core snd_usbmidi_lib bluetooth snd_hwdep intel_rapl_msr snd_rawmidi joydev snd_seq_device videobuf2_v4l2 libarc4 kvm videobuf2_common jitterentropy_rng iwlwifi irqbypass dell_laptop rapl ledtrig_audio intel_cstate drbg videodev dell_wmi iTCO_wdt dell_smm_hwmon intel_pmc_bxt intel_uncore ansi_cprng iTCO_vendor_support snd_pcm dell_smbios dcdbas snd_timer serio_raw wmi_bmof pcspkr dell_wmi_descriptor intel_wmi_thunderbolt snd cfg80211 mc watchdog soundcore processor_thermal_device ecdh_generic intel_pch_thermal hid_multitouch ecc mei_me intel_rapl_common rfkill mei intel_xhci_usb_role_switch roles intel_soc_dts_iosf int3403_thermal evdev int340x_thermal_zone int3400_thermal ac acpi_pad acpi_thermal_rel intel_hid sparse_keymap acpi_cpufreq evdi(OE) parport_pc ppdev sunrpc lp parport ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic algif_skcipher af_alg dm_crypt dm_mod usbhid hid_generic i915 crc32_pclmul crc32c_intel ghash_clmulni_intel
[ 1736.331794]  ahci libahci xhci_pci nvme rtsx_pci_sdmmc xhci_hcd mmc_core libata rtsx_pci i2c_algo_bit nvme_core drm_kms_helper t10_pi crc_t10dif psmouse cec aesni_intel crct10dif_generic drm libaes crypto_simd cryptd usbcore i2c_i801 glue_helper scsi_mod i2c_smbus crct10dif_pclmul crct10dif_common intel_lpss_pci intel_lpss usb_common idma64 fan i2c_hid hid wmi battery video button
[ 1736.331806] CPU: 1 PID: 6113 Comm: insmod Tainted: G         CIOE     5.8.0-0.bpo.2-amd64 #1 Debian 5.8.10-1~bpo10+1
[ 1736.331807] Hardware name: Dell Inc. XPS 13 9350/09JHRY, BIOS 1.12.2 12/15/2019
[ 1736.331809] RIP: 0010:refcount_warn_saturate+0x4e/0xf0
[ 1736.331811] Code: a9 00 00 00 c3 85 f6 74 43 83 fe 01 75 7c 80 3d fc 4a ee 00 00 75 ed 48 c7 c7 60 e2 12 9e c6 05 ec 4a ee 00 01 e8 dd 08 c5 ff <0f> 0b c3 80 3d dc 4a ee 00 00 75 ce 48 c7 c7 88 e2 12 9e c6 05 cc
[ 1736.331811] RSP: 0018:ffff9a9344567b10 EFLAGS: 00010282
[ 1736.331813] RAX: 0000000000000000 RBX: ffff8b58138fb7e8 RCX: 0000000000000027
[ 1736.331813] RDX: 0000000000000027 RSI: ffff8b582f298ac0 RDI: ffff8b582f298ac8
[ 1736.331814] RBP: ffff8b57c3716000 R08: 0000000000000549 R09: 0000000000000004
[ 1736.331815] R10: 0000000000000000 R11: 0000000000000001 R12: ffff8b57f15ce000
[ 1736.331815] R13: ffff8b57f15ce000 R14: ffffffffc1734140 R15: ffff8b57c3716000
[ 1736.331817] FS:  00007f5be9299480(0000) GS:ffff8b582f280000(0000) knlGS:0000000000000000
[ 1736.331818] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1736.331818] CR2: 000055a8b67792d0 CR3: 00000003abdae001 CR4: 00000000003606e0
[ 1736.331819] Call Trace:
[ 1736.331827]  fl2000_device_probe+0x1db/0x250 [fl2000]
[ 1736.331842]  usb_probe_interface+0xe2/0x280 [usbcore]
[ 1736.331847]  really_probe+0x2e0/0x3e0
[ 1736.331850]  driver_probe_device+0xde/0x150
[ 1736.331852]  device_driver_attach+0x4f/0x60
[ 1736.331854]  __driver_attach+0x86/0x140
[ 1736.331856]  ? device_driver_attach+0x60/0x60
[ 1736.331858]  ? device_driver_attach+0x60/0x60
[ 1736.331860]  bus_for_each_dev+0x77/0xc0
[ 1736.331863]  ? klist_add_tail+0x3b/0x70
[ 1736.331865]  bus_add_driver+0x14d/0x1f0
[ 1736.331867]  driver_register+0x6b/0xb0
[ 1736.331878]  usb_register_driver+0x88/0x140 [usbcore]
[ 1736.331880]  ? 0xffffffffc173a000
[ 1736.331883]  do_one_initcall+0x46/0x200
[ 1736.331885]  ? _cond_resched+0x15/0x30
[ 1736.331887]  ? kmem_cache_alloc_trace+0x15d/0x230
[ 1736.331890]  do_init_module+0x5a/0x220
[ 1736.331892]  load_module+0x2335/0x25d0
[ 1736.331895]  ? __kernel_read+0x136/0x160
[ 1736.331897]  ? __do_sys_finit_module+0xa8/0x110
[ 1736.331899]  __do_sys_finit_module+0xa8/0x110
[ 1736.331902]  do_syscall_64+0x44/0xc0
[ 1736.331905]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1736.331906] RIP: 0033:0x7f5be93b3f59
[ 1736.331908] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 07 6f 0c 00 f7 d8 64 89 01 48
[ 1736.331909] RSP: 002b:00007ffc31e73e48 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[... cut ...]
[ 1740.413235] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[ 1740.413237] fl2000_device_probe:Detect interrupt interface.
[ 1740.413240] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[ 1740.413573] usbcore: registered new interface driver fl2000

```
```

sudo lsmod | grep fl2000
fl2000                143360  0
usbcore               315392  18 xhci_hcd,usbnet,snd_usb_audio,usbhid,snd_usbmidi_lib,cdc_mbim,cdc_ncm,usb_storage,cdc_wdm,uvcvideo,btusb,xhci_pci,uas,fl2000
```
```
sudo ./fltest
fl2000 device not connected?
```
Thoughts?

freddale4, 2021-01-29:
@ricodol check if in _/dev/_ there is a file called fl2000-1. 
Try rename to fl2000-0.

`sudo mv fl2000-1 fl2000-0`

I know this works for the sample only.
I'm still struggling with the black screen.

llxiaoyuan, 2021-12-30:
rename https://lore.kernel.org/linux-mm/20200422001422.232330-11-walken@google.com/