FrescoLogic/FL2000 #60: Make error in Slackware 14.2+

tetraetila, 2021-02-04:
bash-5.1# make
make -C /usr/src/linux/ M=/home/tetraetila/Downloads/FL2000-master/src modules
make[1]: Entering directory '/usr/src/linux-5.10.12'
  CC [M]  /home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.o
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |    down_read(&current->mm->mmap_sem);
      |                            ^~~~~~~~
      |                            mmap_base
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:115:26: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |    up_read(&current->mm->mmap_sem);
      |                          ^~~~~~~~
      |                          mmap_base
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:129:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |   down_read(&current->mm->mmap_sem);
      |                           ^~~~~~~~
      |                           mmap_base
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:144:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |   up_read(&current->mm->mmap_sem);
      |                         ^~~~~~~~
      |                         mmap_base
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:228:28: error: too many arguments to function ‘vm_map_ram’
  228 |   surface->mapped_buffer = vm_map_ram(
      |                            ^~~~~~~~~~
In file included from /home/tetraetila/Downloads/FL2000-master/src/fl2000_include.h:30,
                 from /home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:92:14: note: declared here
   92 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:279: /home/tetraetila/Downloads/FL2000-master/src/fl2000_surface.o] Error 1
make[1]: *** [Makefile:1805: /home/tetraetila/Downloads/FL2000-master/src] Error 2
make[1]: Leaving directory '/usr/src/linux-5.10.12'
make: *** [Makefile:42: all] Error 2
bash-5.1# 

bash-5.1# ls -la *.o 
-rw-r--r-- 1 root root 51496 Feb  3 20:36 fl2000_big_table.o
-rw-r--r-- 1 root root  8088 Feb  3 20:36 fl2000_bulk.o
-rw-r--r-- 1 root root 20592 Feb  3 20:36 fl2000_compression.o
-rw-r--r-- 1 root root  1688 Feb  3 20:36 fl2000_desc.o
-rw-r--r-- 1 root root  6088 Feb  3 20:36 fl2000_dev.o
-rw-r--r-- 1 root root 12640 Feb  3 20:36 fl2000_dongle.o
-rw-r--r-- 1 root root  9280 Feb  3 20:36 fl2000_i2c.o
-rw-r--r-- 1 root root 10904 Feb  3 20:36 fl2000_interrupt.o
-rw-r--r-- 1 root root 26536 Feb  3 20:36 fl2000_ioctl.o
-rw-r--r-- 1 root root 12200 Feb  3 20:36 fl2000_module.o
-rw-r--r-- 1 root root 18896 Feb  3 20:36 fl2000_monitor.o
-rw-r--r-- 1 root root  6808 Feb  3 20:36 fl2000_register.o
-rw-r--r-- 1 root root 21080 Feb  3 20:36 fl2000_render.o
bash-5.1#

freddale4, 2021-02-09:
I get the version of Marc-Pierre-Barbier...thank`s Marc!
This version compiled ok on my ubuntu 20.
I enabled Secure Boot to avoid the error "insmod: ERROR: could not insert module fl2000.ko: Operation not permitted"

Now I getting black screen in my external monitor. Just trying to figure out what is the problem now. Thoughts?

sasongko26, 2022-07-07:
> 

Would you share your slackbuilds?