FrescoLogic/FL2000 #61: Not working on Ubuntu 20

thiagocardoso1988, 2021-02-13:
I tried, pointing to my kernel path, but I got an error:

```sh
thiago@thiago-Vostro-5481:~/FL2000/src(master)$ uname -r
5.8.0-43-generic
thiago@thiago-Vostro-5481:~/FL2000/src(master)$ make
make -C /usr/src/linux-headers-`uname -r` M=/home/thiago/FL2000/src modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-43-generic'
  CC [M]  /home/thiago/FL2000/src/fl2000_surface.o
/home/thiago/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/thiago/FL2000/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |    down_read(&current->mm->mmap_sem);
      |                            ^~~~~~~~
      |                            mmap_base
/home/thiago/FL2000/src/fl2000_surface.c:115:26: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |    up_read(&current->mm->mmap_sem);
      |                          ^~~~~~~~
      |                          mmap_base
/home/thiago/FL2000/src/fl2000_surface.c:129:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |   down_read(&current->mm->mmap_sem);
      |                           ^~~~~~~~
      |                           mmap_base
/home/thiago/FL2000/src/fl2000_surface.c:144:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |   up_read(&current->mm->mmap_sem);
      |                         ^~~~~~~~
      |                         mmap_base
/home/thiago/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/thiago/FL2000/src/fl2000_surface.c:228:28: error: too many arguments to function ‘vm_map_ram’
  228 |   surface->mapped_buffer = vm_map_ram(
      |                            ^~~~~~~~~~
In file included from /home/thiago/FL2000/src/fl2000_include.h:30,
                 from /home/thiago/FL2000/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:91:14: note: declared here
   91 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:290: /home/thiago/FL2000/src/fl2000_surface.o] Error 1
make[1]: *** [Makefile:1780: /home/thiago/FL2000/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-43-generic'
make: *** [Makefile:41: all] Error 2
```

I tried building it pointing to another kernel (`linux-headers-5.4.0-59-generic`) and it worked, but I couldn't install the built driver `fl2000.ko`

kenballus, 2021-02-15:
According to [this](https://lkml.org/lkml/2020/5/25/770), `mmap_sem` has been renamed to `mmap_lock`. Do a find and replace replacing `mmap_sem` with `mmap_lock`.

Also, [kernel.org](https://www.kernel.org/doc/htmldocs/kernel-api/API-vm-map-ram.html) says `vm_map_ram` takes 4 arguments, the last of which is the memory protection to use. Passing `PAGE_KERNEL` (like this driver does) is for normal RAM. However, I think the last parameter was removed (judging from the function declaration in the error) so you should try removing the last argument to `vm_map_ram`. Since it's only asking for normal RAM, I think this is probably safe.

After applying those changes, I was able to build this with headers for the `5.10.15-arch1-1` kernel, so I bet it'll build for you too.

esneider29, 2021-02-15:
_esneider@esneider-TUF-Gaming-FX505DT:~/Descargas/FL2000-master/src$ sudo make\
make -C /usr/src/linux-headers-5.8.0-43-generic M= modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.8.0-43-generic'
**make[2]: *** No hay ninguna regla para construir el objetivo 'arch/x86/tools/relocs_32.c', necesario para 'arch/x86/tools/relocs_32.o'.  Alto.**
make[1]: *** [arch/x86/Makefile:211: archscripts] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.8.0-43-generic'
make: *** [Makefile:42: all] Error 2_

@thiagocardoso1988 
with _sudo make_  does not error for mmap_sem... 
but  say error for **_arch/x86/tools/relocs_32.c_**
my uname -r  = 5.8.0-43-generic

GabrielMioranza, 2021-03-23:
@thiagocardoso1988 Como fez para o Ubuntu reconhecer o outro monitor? Aqui mesmo o driver funcionando, o Ubuntu não reconhece a outra tela

ricardojlrufino, 2021-04-20:
> @thiagocardoso1988 Como fez para o Ubuntu reconhecer o outro monitor? Aqui mesmo o driver funcionando, o Ubuntu não reconhece a outra tela

Aparentemente não tem suporte para o Ubuntu, apenas para programação via C, acredito que falta a parte do servidor grafico.

thiagocardoso1988, 2021-05-31:
@kenballus it worked with your advices, but I couldn't manage to install the driver. I'll take a look at it later, but just for keeping it logged here, this is the error I got:

```
insmod: ERROR: could not insert module fl2000.ko: Operation not permitted
```

kenballus, 2021-05-31:
@thiagocardoso1988 You need root permissions to install the kernel module. Try using sudo.

MarcelWaldvogel, 2021-06-11:
I use Ubuntu 21.04 with an FL2000 device, and it seems to work without a driver. `lsusb | grep Fresco` returns
```
Bus 001 Device 012: ID 1d5c:7102 Fresco Logic Generic Billboard Device
```
So [it seems that even though no device driver is configuring alternate mode in time,](http://www.technical-direct.com/en/typec_billboard/) USB-C alternate mode is still automagically configured.

(Sometimes, the HDMI port on the adapter, which is built-in to a *Hama 12in1 USB-C Docking Station*, is not configured; but re-plugging to a different USB-C port on my machine sometimes helps.)

I hope this helps anyone here who has one of those adapters! (Even though the attitude Fresco Logic show in #16 is not really support-worthy.)

BTW: Does anyone know why this (often) works? Or how it can be made more reliable?

BenMcLean981, 2021-06-19:
I think this should be renamed, Im pretty sure this is a kernel 5.x issue, not an Ubuntu 20 issue.

MarcelWaldvogel, 2021-10-31:
@ToniRosa00491 I did not do anything. It somehow just works, even without a driver/patch/configuration. At least on my device (the Hama 12in1), it seems to work right out of the box. This might not work for other FL2000's, though.

(Instead of switching to another USB-C port, when HDMI is not recognized, removing/reconnecting power from the docking station also re-enables HDMI capability. Using that docking station, I can even connect two external monitors, one over HDMI, one over DisplayPort.)