FrescoLogic/FL2000 #63: ERROR: could not insert module fl2000.ko: Invalid module format

GustavoEmmel, 2021-03-31:
running on Ubuntu 14.04 LTS when I run this command:

~/Downloads/FL2000-master/src$ sudo insmod fl2000.ko

insmod: ERROR: could not insert module fl2000.ko: Invalid module format

jrackham21, 2021-08-02:
same, have you been able to fix it?

pedrobritobr, 2021-08-04:
same error here, i'm using Mint 20.2 kernel 5.4.0-58

juanMarulanda, 2021-09-15:
Same here Ubuntu 21.04 linux-headers-5.11.0-34-generic

adnen-med, 2021-12-02:
There is any new update ?

zeuhtof, 2021-12-02:
I have the same problem (Ubuntu 2104 with linux-headers-5.13.0-22-generic)
I have tried unsuccessfully to apply modifications described in pull requests #67 and #70

e-hop, 2022-01-17:
same using kernel 5.11.0-46-generic

carlosedurs, 2022-03-11:
nothing new about it?

shanez1215, 2022-08-06:
> Header mismatch. Install correct header, restart system, make clean and make. it will work.

Good afternoon, this did not work for me. Both the system info and the Makefile use /usr/src/linux-headers-5.15.0-43-generic

shanez1215, 2022-08-06:
UPDATE: Was able to get the module installed by reinstalling my kernel: See https://stackoverflow.com/questions/71746914/linux-kernel-module-development-module-x86-modules-skipping-invalid-relocatio

However, I am now getting the IGNORE monitor plugin event issue