FrescoLogic/FL2000 #64: K/Ubuntu  18.04 Works

ricardojlrufino, 2021-04-20:
Tested with sucess on 
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic

uname -a
Linux ricardo-note 5.4.0-70-generic #78~18.04.1-Ubuntu SMP Sat Mar 20 14:10:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

gcc -v
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)

ricardojlrufino, 2021-04-20:
ONLY SAMPLE in C NOT DISPLAY !!

thiagomneves, 2021-06-10:
Tested unsuccessfully

What did I do wrong?

lsb_release -a
No LSB modules are available.
Distributor ID:	LinuxMint
Description:	Linux Mint 19.3 Tricia
Release:	19.3
Codename:	tricia

uname -a
Linux phoenix-note2 5.4.0-74-generic #83~18.04.1-Ubuntu SMP Tue May 11 16:01:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

gcc -vgcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)