FrescoLogic/FL2000 #73: kernel path not fixed

DriPlu, 2021-06-26:
when i try to compile the kernel driver it gives those errors : 

```
make -C /usr/src/kernels/`uname -r` M= modules
make[1]: *** /usr/src/kernels/5.8.0-59-generic: No such file or directory.  Stop.
make: *** [Makefile:41: all] Error 2

```

could you please fix it ?

LucasFidelis, 2021-06-29:
Hello @DriPlu.
Change from /usr/src/kernels/5.8.0-59-generic to /usr/src/5.8.0-59-generic at the file Makefile.
I hope that help you.

Another suggestion:
At my machine (Ubuntu 20.04), before folder Kernel, I have the prefix linux-headers-(Kernel version) -> (linux-headers-5.8.0-59-generic).
That also should to be change at Makefile.
To see that prefix, use o following command: **ls /usr/src/**

DriPlu, 2021-06-29:
> Hello @DriPlu.
> Change from /usr/src/kernels/5.8.0-59-generic to /usr/src/5.8.0-59-generic at the file Makefile.
> I hope that help you.
> 
> Another suggestion:
> At my machine (Ubuntu 20.04), before folder Kernel, I have the prefix linux-headers-(Kernel version) -> (linux-headers-5.8.0-59-generic).
> That also should to be change at Makefile.
> To see that prefix, use the following command: **ls /usr/src/**

Thanks it works now

closed 2021-06-29 by DriPlu