FrescoLogic/FL2000 #76: [BUG] Not compile in kernel 5.10.0-8-amd64 debian bullseye

JudaRails, 2021-08-04:
**Distributor ID:**	Debian
**Description:**	Debian GNU/Linux 11 (bullseye)
**Release:**	11
**Codename:**	bullseye

**Device identify:**
`Bus 001 Device 014: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter
`

**TRACEBACK:**
> make -C /usr/src/linux-headers-`uname -r` M=/home/Escritorio/FL2000/src modules
> make[1]: se entra en el directorio '/usr/src/linux-headers-5.10.0-8-amd64'
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_module.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_bulk.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_ioctl.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_render.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_dev.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_dongle.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_big_table.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_i2c.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_register.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_monitor.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_desc.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_interrupt.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_compression.o
>   CC [M]  /home/Escritorio/FL2000/src/fl2000_surface.o
> /home/Escritorio/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
> /home/Escritorio/FL2000/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
>   109 |    down_read(&current->mm->mmap_sem);
>       |                            ^~~~~~~~
>       |                            mmap_base
> /home/Escritorio/FL2000/src/fl2000_surface.c:115:26: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
>   115 |    up_read(&current->mm->mmap_sem);
>       |                          ^~~~~~~~
>       |                          mmap_base
> /home/Escritorio/FL2000/src/fl2000_surface.c:129:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
>   129 |   down_read(&current->mm->mmap_sem);
>       |                           ^~~~~~~~
>       |                           mmap_base
> /home/Escritorio/FL2000/src/fl2000_surface.c:144:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
>   144 |   up_read(&current->mm->mmap_sem);
>       |                         ^~~~~~~~
>       |                         mmap_base
> /home/Escritorio/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
> /home/Escritorio/FL2000/src/fl2000_surface.c:228:28: error: too many arguments to function ‘vm_map_ram’
>   228 |   surface->mapped_buffer = vm_map_ram(
>       |                            ^~~~~~~~~~
> In file included from /home/Escritorio/FL2000/src/fl2000_include.h:30,
>                  from /home/Escritorio/FL2000/src/fl2000_surface.c:8:
> /usr/src/linux-headers-5.10.0-8-common/include/linux/vmalloc.h:87:14: note: declared here
>    87 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
>       |              ^~~~~~~~~~
> make[3]: *** [/usr/src/linux-headers-5.10.0-8-common/scripts/Makefile.build:284: /home/Escritorio/FL2000/src/fl2000_surface.o] Error 1
> make[2]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:1845: /home/Escritorio/FL2000/src] Error 2
> make[1]: *** [/usr/src/linux-headers-5.10.0-8-common/Makefile:185: __sub-make] Error 2
> make[1]: se sale del directorio '/usr/src/linux-headers-5.10.0-8-amd64'
> make: *** [Makefile:41: all] Error 2

JudaRails, 2021-08-04:
Compile change fl2000_surface.c
[diff_kernel_5.10.0-8-amd64.txt](https://github.com/FrescoLogic/FL2000/files/6932088/diff_kernel_5.10.0-8-amd64.txt)

emb3dd3d, 2022-01-15:
> Compile change fl2000_surface.c [diff_kernel_5.10.0-8-amd64.txt](https://github.com/FrescoLogic/FL2000/files/6932088/diff_kernel_5.10.0-8-amd64.txt)

very noob here.. how do you apply the diff file  please?? it would be nice to get this going.  I haven't dealt with compiling in years and was a noob then.  any help would be appreciated.

emb3dd3d, 2022-01-15:
> 

nevermind.. googled and found the patch command.  I learned something new.