FrescoLogic/FL2000 #78: Successfully made FLTEST 0, but no Virtual Display is being shown in Ubuntu Settings

luihzpg, 2021-08-17:
After successfully running the FLTEST 0 where it turned the display on with some colored bars, I still can't set the display in Ubuntu since Ubuntu does not show this Display in Display settings. I am using Kernel 5.4.

So why Ubuntu is not recognizing the new display?

Also have to mention that every time that I boot the computer I have to re run the FL2000/src/make and also the sudo insmod fl2000.ko it seems that after I reboot It uninstalls the driver.

dagelf, 2021-09-06:
Make generates the `.ko` file, so once you have that you shouldn't need to `make` it again. Usually `make install` copies it to /lib/modules/* so that the kernel will find it there automatically. You might want to read this: https://tldp.org/LDP/lkmpg/2.6/lkmpg.pdf. 

The manufacturer did not write a full driver, only half of it... this driver does not expose any X display functions. There might be a way to run XVnc on it, or get SDL apps to display directly to it in the mean time.

kelsonc, 2022-01-14:
I have the same problem reported. How to solve?

> After successfully running the FLTEST 0 where it turned the display on with some colored bars, I still can't set the display in Ubuntu since Ubuntu does not show this Display in Display settings. I [...]

I have the same problem reported. How to solve?

Lassebq, 2024-12-17:
Has anyone implemented the display part yet? any fork perhaps? Also, has it been tested on Wayland?