FrescoLogic/FL2000 #84: i got this, anybody can help, please?

trademark007, 2022-03-29:
this make file
```
# Makefile
#
# (c)Copyright 2017, Fresco Logic, Incorporated.
#
# Purpose:
#
# NOTE: DO NOT SEND THIS FILE OUTSIDE OF FRESCO LOGIC.
#

fl2000-y := fl2000_module.o
fl2000-y += fl2000_bulk.o
fl2000-y += fl2000_ioctl.o
fl2000-y += fl2000_render.o
fl2000-y += fl2000_dev.o
fl2000-y += fl2000_dongle.o
fl2000-y += fl2000_big_table.o
fl2000-y += fl2000_i2c.o
fl2000-y += fl2000_register.o
fl2000-y += fl2000_monitor.o
fl2000-y += fl2000_desc.o
fl2000-y += fl2000_interrupt.o
fl2000-y += fl2000_compression.o
fl2000-y += fl2000_surface.o
fl2000-y += fl2000_fops.o
fl2000-y += fl2000_hdmi.o

ifdef CONFIG_USB_FL2000

obj-$(CONFIG_USB_FL2000) := fl2000.o

else

obj-m := fl2000.o

#KERNEL_PATH = ../../../kernel/linux-3.10.0-327.13.1.el7
#KERNEL_PATH = /usr/src/kernels/3.10.0-327.13.1.el7.x86_64
# KERNEL_PATH = /usr/src/kernels/`uname -r`
KERNEL_PATH = /usr/src/linux-headers-`uname -r`

all:
	make -C $(KERNEL_PATH) M=$(PWD) modules

clean:
	make -C $(KERNEL_PATH) M=$(PWD) clean
	rm -f Module.symvers

endif

```

this following error
```
make -C /usr/src/linux-headers-`uname -r` M=/home/smileydox/Downloads/FL2000-master/src modules
make[1]: Entering directory '/usr/src/linux-headers-5.14.0-kali2-amd64'
  CC [M]  /home/smileydox/Downloads/FL2000-master/src/fl2000_surface.o
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:109:28: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |    down_read(&current->mm->mmap_sem);
      |                            ^~~~~~~~
      |                            mmap_base
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:115:26: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |    up_read(&current->mm->mmap_sem);
      |                          ^~~~~~~~
      |                          mmap_base
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:129:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |   down_read(&current->mm->mmap_sem);
      |                           ^~~~~~~~
      |                           mmap_base
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:144:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |   up_read(&current->mm->mmap_sem);
      |                         ^~~~~~~~
      |                         mmap_base
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:228:28: error: too many arguments to function ‘vm_map_ram’
  228 |   surface->mapped_buffer = vm_map_ram(
      |                            ^~~~~~~~~~
In file included from /home/smileydox/Downloads/FL2000-master/src/fl2000_include.h:30,
                 from /home/smileydox/Downloads/FL2000-master/src/fl2000_surface.c:8:
/usr/src/linux-headers-5.14.0-kali2-common/include/linux/vmalloc.h:126:14: note: declared here
  126 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-5.14.0-kali2-common/scripts/Makefile.build:276: /home/smileydox/Downloads/FL2000-master/src/fl2000_surface.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.14.0-kali2-common/Makefile:1882: /home/smileydox/Downloads/FL2000-master/src] Error 2
make[1]: *** [/usr/src/linux-headers-5.14.0-kali2-common/Makefile:232: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-kali2-amd64'
make: *** [Makefile:41: all] Error 2

```

isabekov, 2022-05-20:
Check this out: https://github.com/isabekov/FL2000/commit/5ef4526d33adc00a4e4ad6cedd85bf88640354b4
It implements the patch submitted by user **hcar**  in the AUR repo of the driver.
I was able to compile the driver for kernel version 5.14.12:
   
    $  lsusb | grep Fresco
        Bus 002 Device 002: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter

    $ sudo insmod  /lib/modules/5.14.12-arch1-1/extramodules/fl2000.ko.zst

    $ sudo modinfo fl2000
    filename:       /lib/modules/5.14.12-arch1-1/extramodules/fl2000.ko.zst
    license:        GPL
    description:    USB VGA device driver - Version 0.0.0.1
    author:         Fresco Logic
    srcversion:     3B73451892926650E860085
    alias:          usb:v1D5Cp2000d*dc*dsc*dp*ic*isc02ip*in*
    depends:
    retpoline:      Y
    name:           fl2000
    vermagic:       5.14.12-arch1-1 SMP preempt mod_unload

Do not forget to compile `fltest` program in directory `FL2000/sample`  by executing:

    $ make fltest

and testing the driver:

    $ sudo ./fltest 0

It will display a couple of images on the monitor connected to the adapter.

There is no support for Xorg, so the driver does not have any practical use within a desktop environment.

trademark007, 2022-05-25:
thanks for your suggest it works for the first time until i got this error when try to reconnect again
```
fl2000 device not connected?
```
so, look at this issues https://github.com/FrescoLogic/FL2000/issues/31#issuecomment-546146187 but still doesnt work too

conbon-boncon, 2022-07-26:
Function mmap_sem is deprecated. Use mmap_lock. Replace down_read and up_read with this

`down_read(&current->mm->mmap_sem);
mmap_read_lock(current->mm);`

Also, vm_map_ram was changed. It takes one less parameter. Remove parameter PAGE_KERNEL