FrescoLogic/FL2000 #86: Kernel panic after install driver RHEL 7.6

alexleosumy, 2022-11-08:
Hello. I get kernel panic and reboot PC after insmod command.
`uname -a
Linux hostname 3.10.0-1160.66.1.el7.x86_64 #1 SMP Wed Apr 27 20:34:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
What data do you need to solve this issue?

alexleosumy, 2022-11-08:
`[ 1878.402062] fl2000_device_probe:bInterfaceNumber:1 bAlternateSetting:0
[ 1878.402067] ------------[ cut here ]------------
[ 1878.402076] WARNING: CPU: 4 PID: 11229 at include/linux/kref.h:52 fl2000_device_probe+0x304/0x3e0 [fl2000]
[ 1878.402078] Modules linked in: fl2000(OE+) tcp_lp nvidia_uvm(OE) lp parport xt_nat veth nf_conntrack_netlink xt_addrtype br_netfilter xt_CHECKSUM ipt_MASQUERADE nf_nat_masquerade_ipv4 tun bridge stp llc devlink ebtable_filter ebtables ip6_tables overlay(T) nf_conntrack_ftp ipt_REJECT nf_reject_ipv4 xt_conntrack iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle sunrpc nvidia_drm(POE) nvidia_modeset(POE) vfat fat nvidia(POE) ext4 mbcache jbd2 iTCO_wdt iTCO_vendor_support skx_edac intel_powerclamp coretemp intel_rapl iosf_mbi kvm_intel kvm irqbypass crc32_pclmul ghash_clmulni_intel aesni_intel lrw gf128mul joydev glue_helper ablk_helper cryptd sg i2c_i801 pcspkr mei_me mei ipmi_si ipmi_devintf ipmi_msghandler acpi_power_meter acpi_pad tpm_crb
[ 1878.402127]  binfmt_misc ip_tables xfs libcrc32c sd_mod crc_t10dif crct10dif_generic cdc_ether usbnet mii mgag200 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ttm ahci i40e drm libahci igb crct10dif_pclmul crct10dif_common crc32c_intel libata nvme nvme_core dca ptp pps_core i2c_algo_bit drm_panel_orientation_quirks wmi nfit libnvdimm uas usb_storage dm_mirror dm_region_hash dm_log dm_mod fuse
[ 1878.402159] CPU: 4 PID: 11229 Comm: insmod Kdump: loaded Tainted: P           OE  ------------ T 3.10.0-1160.66.1.el7.x86_64 #1
[ 1878.402161] Hardware name: Lenovo ThinkSystem SE350 -[7D1XCTO1WW]-/-[7D1XCTO1WW]-, BIOS -[HYE122F-1.19]- 04/27/2022
[ 1878.402163] Call Trace:
[ 1878.402171]  [<ffffffff953865a9>] dump_stack+0x19/0x1b
[ 1878.402179]  [<ffffffff94c9b408>] __warn+0xd8/0x100
[ 1878.402184]  [<ffffffff94c9b54d>] warn_slowpath_null+0x1d/0x20
[ 1878.402190]  [<ffffffffc2f6e504>] fl2000_device_probe+0x304/0x3e0 [fl2000]
[ 1878.402198]  [<ffffffff951199ed>] usb_probe_interface+0x11d/0x320
[ 1878.402204]  [<ffffffff950bbe85>] driver_probe_device+0xc5/0x3e0
[ 1878.402208]  [<ffffffff950bc283>] __driver_attach+0x93/0xa0
[ 1878.402212]  [<ffffffff950bc1f0>] ? __device_attach+0x50/0x50
[ 1878.402216]  [<ffffffff950b9a25>] bus_for_each_dev+0x75/0xc0
[ 1878.402220]  [<ffffffff950bb7fe>] driver_attach+0x1e/0x20
[ 1878.402224]  [<ffffffff950bb2a0>] bus_add_driver+0x200/0x2d0
[ 1878.402228]  [<ffffffff950bc914>] driver_register+0x64/0xf0
[ 1878.402232]  [<ffffffff95118052>] usb_register_driver+0x82/0x160
[ 1878.402238]  [<ffffffffc2f8d000>] ? 0xffffffffc2f8cfff
[ 1878.402244]  [<ffffffffc2f8d01e>] init_module+0x1e/0x1000 [fl2000]
[ 1878.402249]  [<ffffffff94c0210a>] do_one_initcall+0xba/0x240
[ 1878.402256]  [<ffffffff94d1ea5a>] load_module+0x271a/0x2bb0
[ 1878.402262]  [<ffffffff94fb4ce0>] ? ddebug_proc_write+0x100/0x100
[ 1878.402267]  [<ffffffff94e50b83>] ? fput+0x13/0x20
[ 1878.402271]  [<ffffffff94d1a5e3>] ? copy_module_from_fd.isra.44+0x53/0x150
[ 1878.402276]  [<ffffffff94d1f0d6>] SyS_finit_module+0xa6/0xd0
[ 1878.402281]  [<ffffffff95399f92>] system_call_fastpath+0x25/0x2a
[ 1878.402284] ---[ end trace 5efb6a27c6074aea ]---
[ 1878.402286] fl2000_device_probe:streaming interface detected
[ 1878.402434] fl2000_device_probe:usb_dev(ffff9cd398d00400), minor_num(128), dev_name(fl2000-0) created
[ 1879.158498] fl2000_monitor_vga_status_handler:Ignore MonitorPlugout event, monitor not attached.
[ 1879.158534] fl2000_device_probe:bInterfaceNumber:2 bAlternateSetting:0
[ 1879.158536] fl2000_device_probe:Detect interrupt interface.
[ 1879.158538] fl2000_intr_pipe_create:found ep_num_intr_in(3)
[ 1879.158708] usbcore: registered new interface driver fl2000
[ 1928.560902] NMI watchdog: Watchdog detected hard LOCKUP on cpu 9`

Tnology, 2023-04-01:
Did you ever get this resolved?