FrescoLogic/FL2000 #95: How to fix the file fl2000_surface.c after use command make in 5.19.0-46-generic

fusesuphasin, 2023-07-17:
make -C /usr/src/linux-headers-5.19.0-45-generic M=/home/suphasin/FL2000/src modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-45-generic'
  CC [M]  /home/suphasin/FL2000/src/fl2000_surface.o
/home/suphasin/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/suphasin/FL2000/src/fl2000_surface.c:109:49: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |                         down_read(&current->mm->mmap_sem);
      |                                                 ^~~~~~~~
      |                                                 mmap_base
/home/suphasin/FL2000/src/fl2000_surface.c:115:47: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |                         up_read(&current->mm->mmap_sem);
      |                                               ^~~~~~~~
      |                                               mmap_base
/home/suphasin/FL2000/src/fl2000_surface.c:129:41: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |                 down_read(&current->mm->mmap_sem);
      |                                         ^~~~~~~~
      |                                         mmap_base
/home/suphasin/FL2000/src/fl2000_surface.c:144:39: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |                 up_read(&current->mm->mmap_sem);
      |                                       ^~~~~~~~
      |                                       mmap_base
/home/suphasin/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/suphasin/FL2000/src/fl2000_surface.c:228:42: error: too many arguments to function ‘vm_map_ram’
  228 |                 surface->mapped_buffer = vm_map_ram(
      |                                          ^~~~~~~~~~
In file included from /home/suphasin/FL2000/src/fl2000_include.h:30,
                 from /home/suphasin/FL2000/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:129:14: note: declared here
  129 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:257: /home/suphasin/FL2000/src/fl2000_surface.o] Error 1
make[1]: *** [Makefile:1857: /home/suphasin/FL2000/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-45-generic'
make: *** [Makefile:41: all] Error 2

______________________________
Then i change mmap_sem to mmap base then its still error with pointer

so i fixed like this issue, its still not run pass yet.
https://github.com/FrescoLogic/FL2000/issues/94

D-u-c-k-s-e-l, 2023-08-10:
I have the same issues with `5.15.0-78-generic`.

Here's my output:
```
make -C /usr/src/linux-headers-`uname -r` M=/home/ducksel/Downloads/FL2000-master/src modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-78-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
  You are using:           gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_module.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_bulk.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_ioctl.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_render.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_dev.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_dongle.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_big_table.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_i2c.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_register.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_monitor.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_desc.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_interrupt.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_compression.o
  CC [M]  /home/ducksel/Downloads/FL2000-master/src/fl2000_surface.o
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:109:49: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |                         down_read(&current->mm->mmap_sem);
      |                                                 ^~~~~~~~
      |                                                 mmap_base
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:115:47: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |                         up_read(&current->mm->mmap_sem);
      |                                               ^~~~~~~~
      |                                               mmap_base
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:129:41: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |                 down_read(&current->mm->mmap_sem);
      |                                         ^~~~~~~~
      |                                         mmap_base
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:144:39: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |                 up_read(&current->mm->mmap_sem);
      |                                       ^~~~~~~~
      |                                       mmap_base
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:228:42: error: too many arguments to function ‘vm_map_ram’
  228 |                 surface->mapped_buffer = vm_map_ram(
      |                                          ^~~~~~~~~~
In file included from /home/ducksel/Downloads/FL2000-master/src/fl2000_include.h:30,
                 from /home/ducksel/Downloads/FL2000-master/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:133:14: note: declared here
  133 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
make[2]: *** [scripts/Makefile.build:297: /home/ducksel/Downloads/FL2000-master/src/fl2000_surface.o] Error 1
make[1]: *** [Makefile:1914: /home/ducksel/Downloads/FL2000-master/src] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-78-generic'
make: *** [Makefile:41: all] Error 2
```

5hun5uke, 2023-08-19:
This is fixed by #67