FrescoLogic/FL2000 #97: Error with make

Manuelcarvajal09, 2023-08-30:
Hi i'm new Fedora user and when i use make the terminal say:
```shell
 manuel   master U:2 ?:29  ~/FL2000/src  make
make -C /usr/src/kernels/6.4.12-200.fc38.x86_64 M=/home/manuel/FL2000/src modules
make[1]: Entering directory '/usr/src/kernels/6.4.12-200.fc38.x86_64'
  CC [M]  /home/manuel/FL2000/src/fl2000_surface.o
/home/manuel/FL2000/src/fl2000_surface.c: In function ‘fl2000_get_user_pages’:
/home/manuel/FL2000/src/fl2000_surface.c:60:28: error: ‘FOLL_TOUCH’ undeclared (first use in this function); did you mean ‘FOLL_FORCE’?
   60 |                 FOLL_GET | FOLL_TOUCH,
      |                            ^~~~~~~~~~
      |                            FOLL_FORCE
/home/manuel/FL2000/src/fl2000_surface.c:60:28: note: each undeclared identifier is reported only once for each function it appears in
/home/manuel/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_pin_down’:
/home/manuel/FL2000/src/fl2000_surface.c:109:49: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  109 |                         down_read(&current->mm->mmap_sem);
      |                                                 ^~~~~~~~
      |                                                 mmap_base
/home/manuel/FL2000/src/fl2000_surface.c:115:47: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  115 |                         up_read(&current->mm->mmap_sem);
      |                                               ^~~~~~~~
      |                                               mmap_base
/home/manuel/FL2000/src/fl2000_surface.c:129:41: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  129 |                 down_read(&current->mm->mmap_sem); mmap_read_lock(current->mm);
      |                                         ^~~~~~~~
      |                                         mmap_base
/home/manuel/FL2000/src/fl2000_surface.c:137:31: error: assignment of read-only member ‘vm_flags’
  137 |                 vma->vm_flags &= ~(VM_IO | VM_PFNMAP);
      |                               ^~
/home/manuel/FL2000/src/fl2000_surface.c:143:31: error: assignment of read-only member ‘vm_flags’
  143 |                 vma->vm_flags = old_flags;
      |                               ^
/home/manuel/FL2000/src/fl2000_surface.c:144:39: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  144 |                 up_read(&current->mm->mmap_sem);
      |                                       ^~~~~~~~
      |                                       mmap_base
/home/manuel/FL2000/src/fl2000_surface.c: In function ‘fl2000_surface_map’:
/home/manuel/FL2000/src/fl2000_surface.c:228:42: error: too many arguments to function ‘vm_map_ram’
  228 |                 surface->mapped_buffer = vm_map_ram(
      |                                          ^~~~~~~~~~
In file included from /home/manuel/FL2000/src/fl2000_include.h:30,
                 from /home/manuel/FL2000/src/fl2000_surface.c:8:
./include/linux/vmalloc.h:131:14: note: declared here
  131 | extern void *vm_map_ram(struct page **pages, unsigned int count, int node);
      |              ^~~~~~~~~~
/home/manuel/FL2000/src/fl2000_surface.c: In function ‘fl2000_get_user_pages’:
/home/manuel/FL2000/src/fl2000_surface.c:65:1: error: control reaches end of non-void function [-Werror=return-type]
   65 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /home/manuel/FL2000/src/fl2000_surface.o] Error 1
make[1]: *** [Makefile:2050: /home/manuel/FL2000/src] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.4.12-200.fc38.x86_64'
make: *** [Makefile:41: all] Error 2

``` 
How i can solve this? i don't know how to modify the surface file.

thomas-delplace, 2024-04-06:
I've been encountering the exact same issue with make. I'd love to have a solution too. :)

freddale4, 2024-04-12:
Try to run make command with SUDO