FrescoLogic/FL2000 #99: VGA with VGA/USB 3.0 to USB 2.0 !!

Leogiamminola, 2025-01-04:
Hello everyone!
Thank you in advance for the space to share and solve the problems and for your time in reading them.

I'm just starting out in programming and I'm trying something (which I understand isn't possible at the moment) but I'm sharing my doubts in case there's someone else out there with the same problem (or solution).

I am trying to connect an extra 3rd screen (in extension mode) to my HP 15ay notebook with Kali Linux as operating system (it only has one HDMI port to which I have a 2nd screen connected that works correctly and I only have USB 2.0 ports available...It also doesn't have USB C ports), via a VGA/USB 3.0 connector, knowing their incompatibilities and limitations...

The adapter is  (lsusb) "Bus 002 Device 002: ID 1d5c:2000 Fresco Logic FL2000/FL2000DX VGA/DVI/HDMI Adapter" (which I understand is compatible with Kali Linux but I have problems with the drivers).

I installed the Driver for Fresco Logic FL2000, downloaded and installed the necessary dependencies

sudo apt update
sudo apt install git build-essential dkms linux-headers-linux-headers-6.11.2-rt-amd64

I cloned the driver repository

git clone https://github.com/FrescoLogic/FL2000.git

but it doesn't recognize the monitor and I can't compile...

I installed and edited the Kernel headers and the "Makefile" file

nano Makefile
KERNEL_PATH = /usr/src/linux-headers-linux-headers-6.11.2-rt-amd64

and when compiling the Kernel module and trying to load the module by running "make", it does not generate the file called fl2000.ko.
I understand It is necessary to adjust the driver code to be compatible with my kernel version...

After being able to solve almost all the errors (EXCEPT ONE!) with the help of chatgpt (such as problems associated with the get_user_pages, mmap_sem, vm_map_ram functions), modifying the fl2000_surface.c file, I understand that the problem I need to solve is with the "flags" (vm_flags) that are not defined or available in my kernel version...yet!

I wanted to know if there is an updated version of the driver code (FL2000) compatible with my kernel (since I don't want to compile it with an older version..)

So this is the error I get when running "make":

home/kali/FL2000/src/fl2000_fops.c: In function ‘set_vm_flags’:
/home/kali/FL2000/src/fl2000_fops.c:136:19: error: assignment of read-only member ‘vm_flags’
  136 |     vma->vm_flags = old_flags | flags;
      |                   ^
/home/kali/FL2000/src/fl2000_fops.c: In function ‘clear_vm_flags’:
/home/kali/FL2000/src/fl2000_fops.c:141:19: error: assignment of read-only member ‘vm_flags’
  141 |     vma->vm_flags = old_flags & ~flags;
      |                   ^
make[3]: *** [/usr/src/linux-headers-6.11.2-common-rt/scripts/Makefile.build:249: /home/kali/FL2000/src/fl2000_fops.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.11.2-common-rt/Makefile:1951: /home/kali/FL2000/src] Error 2
make[1]: *** [/usr/src/linux-headers-6.11.2-common-rt/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.11.2-rt-amd64'
make: *** [Makefile:42: all] Error 2

And this is how I got the file "fl2000_fops.c :

// fl2000_fops.c
//
// (c)Copyright 2017, Fresco Logic, Incorporated.
//
// Purpose:
//

#include "fl2000_include.h"

#define USE_VM_INSERT_PFN       0
 
int fl2000_open(struct inode * inode, struct file * file)
{
        int const minor = iminor(inode);
        struct usb_interface * interface;
        struct dev_ctx * dev_ctx;
        int ret_val;
        uint32_t open_count;

        ret_val = 0;
[... 159 lines cut ...]

    // Continuar con el resto de la función
    ...
    return 0;
}

    for (i = 0; i < num_pages; i++) {
        unsigned long usr_addr = vma->vm_start + (i << PAGE_SHIFT);
        struct page *page = dev_ctx->start_page + i;

        ret_val = vm_insert_page(vma, usr_addr, page);
        if (ret_val) {
            pr_err("vm_insert_page(usr_addr 0x%lx), page_count(%u) failed %>
                   usr_addr, page_count(page), ret_val);
            break;
        }
    }
pr_info("vm_flags(0x%lx)\n", vma->vm_flags);
    return ret_val;
}

 
Thank you very much in advance and I hope the problem can be understood and I would appreciate the solution just to know that it is possible!

Otherwise I will buy a DockStation or another PC, which is what I am trying to avoid, haha, thanks, greetings!

CanNuhlar, 2025-01-04:
vm_flags are read only on kernel 6.x. There are wrapper functions called
vm_flags_set and vm_flags_clear. You need to rewrite it like this;

If you see "vma->vm_flags = old_flags | flags"  this translates to
"vm_flags_set(vma, old_flags | flags)"
If you see "vma->vm_flags = old_flags & ~flags" this translates to "
vm_flags_clear(vma, flags); vm_flags_set(vma, old_flags & ~flags); "

Leo Giamminola ***@***.***>, 4 Oca 2025 Cmt, 19:29 tarihinde
şunu yazdı:

CanNuhlar, 2025-01-04:
Just a remark: Even if you manage to compile this driver for your kernel
version, this driver lacks the DRM support which means you won't be able to
use it as a monitor. Rather a device that you can send bitmap info to show
something on the hooked up display. This is not something for end users.

Leo Giamminola ***@***.***>, 4 Oca 2025 Cmt, 19:29 tarihinde
şunu yazdı:

Leogiamminola, 2025-01-04:
:/  Ok I understand, then I will look for other possible solutions 
thank you very much for the prompt and accurate response, 
Thank you for your time and knowledge 😌​🙏​

ulli-kroll, 2025-01-25:
I hope this helps, to understand the *multiple* issues, 
on the hardware side with this deivce

https://github.com/FrescoLogic/FL2000/issues/12

At this time, I *had* a working prov of DRM

The biggest limiting part is bandwith